betaflight/betaflight-configurator #871: To discuss: package-lock.json

McGiverGim, 2018-01-18:
With the latest PR some "versioning" problem has appear.

The idea under `npm` is that the dependencies are being declared in `package.json`, the `npm install` generates a `package-lock.json` with detailed dependencies, this file is uploaded to the repository and since then any `npm install` will use the `package-lock.json`:  in this way all builds will have the same dependencies.
Problems:
- If we modify the `package.json` and we don't upload a new `package-lock.json` then the `npm install` will create the lock file with maybe different dependencies than the original.
- We have one dependency by system. The `appdmg` is only available in OSX. So the lock file for OSX and others will be different. We can move the package to the "general" list, but we will need to use an `npm install --force` to not get an error (and a force usually is not a good idea).

So the question is, what we do with this dependency? We must upload a `package-lock.json` for windows and linux but not for osx? We must use the `--force`?

McGiverGim, 2018-01-18:
We use the dependencies in an "open" way, using ~ and ^ to let `npm` adapt the version to the inner dependencies found between all the packages included.

This makes that, for example, if I build it now, I can get some dependencies, but if you do it tomorrow, your version can have other dependencies.

This can lead to problems, for example: https://github.com/betaflight/betaflight-configurator/pull/868

The `package-lock.json` is the one that has the real dependencies and versions.

mikeller, 2018-01-18:
Not generating or persisting the `package-lock.json` file is contrary to what we want to achieve - this means dependency versions will be re-evaluated with every build (local and on CI), and it will be impossible to get repeatable builds.

> If we modify the package.json and we don't upload a new package-lock.json then the npm install will create the lock file with maybe different dependencies than the original.

This should never happen. Any update to `package.json` has to be followed by an `npm install`, triggering an update to `package-lock.json`, and both `package.json` and `package-lock.json` have to be part of the pull request.

In the way that we are set up now, there is one exception to this: Because the `appdmg` package only exists for macos, we are installing it through `platform-dependent-modules`, which runs as a `post-install` step and installs `appdmg` if it's running on macos. This will update both `package.json` and `package-lock.json`, and add `appdmg` to them, making them unusable on non-macosx. I don't know how to best handle this, but pushing the output of `npm install` on macosx into the repository is not the solution.

The real reason for problems like #868 is that, because we are using a lot of packages that are designed to only run on a specific architecture, any change to the build system (be it a dependency update or some other change to the setup) can easily break the build for any of the platforms. Unless every change is tested on all 3 platforms before a pull request is merged, this will not be detected before the change is in master. The only way to avoid this will be to set up a CI system like travis for the firmware, that does not only build master, but also builds all pull requests for all platforms. (This situation is not unlike what we already got for the different targets of the firmware.)

basdelfos, 2018-01-18:
One solution would be to remove the `appdmg` from the `platform-dependent-modules` and add this as a manual installation step in the instructions. This way we can include `package.json` and `package-lock.json` in PRs that are created on a mac.

mikeller, 2018-01-18:
@basdelfos: That will make it impossible to do automated builds for Mac on CI...

Another option would be to add `appdmg` as an optional dependency. This will cause the package to not be installed on non - macos with a warning. The drawback is that this will add it to the release for macosx. But that's just a minor drawback.

What do you think?

basdelfos, 2018-01-18:
Oh yeah, didn't think about the automated builds.
Your option could work and is possibly the best work-around.
> The drawback is that this will add it to the release for macosx. But that's just a minor drawback.

What do you mean?

mikeller, 2018-01-18:
As a (non-dev) dependency, it will get installed into the `dist/` folder...

basdelfos, 2018-01-18:
Oh really, that's a drawback. We need to exclude this when coping file source files from `dist/` folder when creating the apps, else all (all platforms) apps wil include the files from `appdmg`. Or - more clean - use another folder name than `dist/` when copying the files for distribution.

mikeller, 2018-01-18:
> else all (all platforms) apps wil include the files from appdmg

No, only macos - it does not install for any other platform.

 > use another folder name than dist/ when copying the files for distribution.

The whole point of the dependencies is to have them installed in the `dist/` folder - if we move the folder where we assemble the distribution we'll have to move the target folder for the dependencies as well.

basdelfos, 2018-01-18:
OK - but we don't use any of the npm packages into the actual app source code (at the moment)

mikeller, 2018-01-18:
Yeah - at the moment we can just not install the dependencies. For the future we should try to find a solution that allows us to install dependencies **and** use `appdmg`.

closed 2018-05-13 by stale[bot]