betaflight/betaflight-configurator #873: Gulp TypeError: Cannot read property 'apply' of undefined

richard-scott, 2018-01-18:
I have a script that runs when I start my laptop which builds the latest configurator from Github:

```
@echo off
cd %USERPROFILE%\
if not exist "Git\" mkdir Git
cd Git
if not exist "betaflight-configurator\" call git clone --depth 1 https://github.com/betaflight/betaflight-configurator.git
cd betaflight-configurator
call git reset --hard
call git pull
if not exist "node_modules\" call npm install && call npm install npm && call npm install gulp
call npm update && call npm update npm && call npm update gulp
call gulp apps --win64
```

As of today, this is now reporting the following error:

```
PS C:\users\micro\Git> .\build-betaflight-configurator.cmd
Cloning into 'betaflight-configurator'...
remote: Counting objects: 422, done.
remote: Compressing objects: 100% (351/351), done.
remote: Total 422 (delta 88), reused 259 (delta 61), pack-reused 0R
Receiving objects: 100% (422/422), 8.40 MiB | 1.70 MiB/s, done.
Resolving deltas: 100% (88/88), done.
HEAD is now at 7a6ae52 Merge pull request #872 from mikeller/update_package_lock
Already up-to-date.

> betaflight-configurator@10.1.0 postinstall C:\Users\micro\Git\betaflight-configurator
> npm run _postinstall

> betaflight-configurator@10.1.0 _postinstall C:\Users\micro\Git\betaflight-configurator
> node ./node_modules/platform-dependent-modules/cli.js

* [platform-dependent-modules] process.cwd(): C:\Users\micro\Git\betaflight-configurator __dirname: C:\Users\micro\Git\betaflight-configurator\node_modules\platform-dependent-modules\lib
* [platform-dependent-modules] No packages to install, skipping.
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@1.1.3 (node_modules\fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.1.3: wanted {"os":"darwin","arch":"any"} (current: {"os":"win32","arch":"x64"})

added 801 packages in 22.178s
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@1.1.3 (node_modules\fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.1.3: wanted {"os":"darwin","arch":"any"} (current: {"os":"win32","arch":"x64"})

+ npm@5.6.0
added 592 packages in 34.382s
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@1.1.3 (node_modules\fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.1.3: wanted {"os":"darwin","arch":"any"} (current: {"os":"win32","arch":"x64"})

+ gulp@4.0.0
added 116 packages and updated 1 package in 20.732s
Building for platform(s): win64.
[11:45:40] Using gulpfile ~\Git\betaflight-configurator\gulpfile.js
C:\Users\micro\AppData\Roaming\npm\node_modules\gulp\bin\gulp.js:129
    gulpInst.start.apply(gulpInst, toRun);
                   ^

TypeError: Cannot read property 'apply' of undefined
    at C:\Users\micro\AppData\Roaming\npm\node_modules\gulp\bin\gulp.js:129:20
    at _combinedTickCallback (internal/process/next_tick.js:131:7)
    at process._tickCallback (internal/process/next_tick.js:180:9)
    at Function.Module.runMain (module.js:686:11)
    at startup (bootstrap_node.js:187:16)
    at bootstrap_node.js:608:3
PS C:\users\micro\Git>
```

The last time this was working ok was Monday or Tuesday evening...
Am I using the correct build process?

For reference, the same build process works fine for the blheli-configurator.

mikeller, 2018-01-18:
That is caused by the update of gulp to version 4: #837.

You'll have to:
- remove the global installation of `gulp`;
- install the latest version of `gulp-cli` (>= 2.0.0).

McGiverGim, 2018-01-18:
Your script, if I'm not wrong only calls `npm install` when the `node_modules` does not exist. If not, it calls `npm udpate` only. Maybe is better to call the `npm install` always: https://stackoverflow.com/questions/12478679/npm-install-vs-update-whats-the-difference

richard-scott, 2018-01-18:
now I get this:
```
PS C:\users\micro\Git\betaflight-configurator> gulp --version
[14:29:44] CLI version 2.0.0
[14:29:44] Local version 3.9.1
PS C:\users\micro\Git\betaflight-configurator> gulp apps
Building for platform(s): win32.
C:\users\micro\Git\betaflight-configurator\gulpfile.js:43
gulp.task('clean', gulp.parallel(clean_dist, clean_apps, clean_debug, clean_release));
                        ^

TypeError: gulp.parallel is not a function
    at Object.<anonymous> (C:\users\micro\Git\betaflight-configurator\gulpfile.js:43:25)
    at Module._compile (module.js:660:30)
    at Object.Module._extensions..js (module.js:671:10)
    at Module.load (module.js:573:32)
    at tryModuleLoad (module.js:513:12)
    at Function.Module._load (module.js:505:3)
    at Module.require (module.js:604:17)
    at require (internal/module.js:11:18)
    at execute (C:\Users\micro\AppData\Roaming\npm\node_modules\gulp-cli\lib\versioned\^3.7.0\index.js:28:18)
    at Liftoff.handleArguments (C:\Users\micro\AppData\Roaming\npm\node_modules\gulp-cli\index.js:175:63)
```

richard-scott, 2018-01-18:
ah, cancel that. I wasn't using the 9.4.0 "current" version of NodeJS but the 8.9.4 LTS version.
Build is working again now. Thanks for all the help.

closed 2018-01-18 by richard-scott

richard-scott, 2018-01-18:
After mucking about all day, I realised I no longer need "npm install npm" and I ended up with this as my build script:
```
@echo off
cd %USERPROFILE%\
if not exist "Git\" mkdir Git
cd Git
if not exist "betaflight-configurator\" call git clone --depth 1 https://github.com/betaflight/betaflight-configurator.git
cd betaflight-configurator
call git reset --hard
call git pull
call npm install
call gulp apps
```

It works the same for the blackbox-log-viewer and blheli-configurator Standalone Apps too :-)