betaflight/betaflight-configurator #874: GPS Position is not shown

jstremmler, 2018-01-19:
In the new Betaflight Configurator 10.1. GPS position is not shown on the GPS Page anymore
[image]

McGiverGim, 2018-01-19:
Some time ago this problem happened and was solved, I can find the BF PR but I found the CF one: https://github.com/cleanflight/cleanflight-configurator/pull/485 

Maybe the NW is ignoring the `manifest.js`? Can you build yourself the configurator in debug mode and take a look to the console to see if there's some error?

McGiverGim, 2018-01-19:
I've tried and this is the error in the console:
```
extensions::webViewEvents:294 <webview>: The load has aborted with error -20: ERR_BLOCKED_BY_CLIENT.
```

McGiverGim, 2018-01-19:
I think that I've fixed it in https://github.com/betaflight/betaflight-configurator/pull/875, only need is to declare the partition.

mikeller, 2018-01-19:
> Maybe the NW is ignoring the manifest.js? Can you build yourself the configurator in debug mode and take a look to the console to see if there's some error?

Yes, NW does not use `manifest.json`, and gets the metadata from `package.json` instead.

closed 2018-01-19 by mikeller