betaflight/betaflight-configurator #880: Installation Time

coppernanotube, 2018-01-21:
I downloaded the latest version(10.10) of the stand-alone windows installer and it is taking ages to install, I am sitting at around 45 minutes and it is still searching my entire hard drive for some reason (I assume looking for an old install). This is a rather large drive so it is bogging down the install process. It is currently checking every single brush file in my Gimp 2.8 folder which I assume isn't the intended behaviour for the installer. 

Thanks for the help, and for the great software. 

Cheers.

McGiverGim, 2018-01-21:
No, the installer uses the registry to locate all installs. In what screen of the installer is it stopped?

coppernanotube, 2018-01-21:
Here is a screenshot of the current progress, it seems to be searching through my Dolphin folder now....

[image]

McGiverGim, 2018-01-21:
The problem is that you have selected the program files as output folder, without creating a subfolder inside. It checks the existing files to not remove them when uninstalling.

coppernanotube, 2018-01-21:
Ah, that makes sense, it installed in about 30 seconds with its own sub-folder. 

Thanks.

McGiverGim, 2018-01-21:
Yeah, we need to add a verification to check that the destination folder is empty, for the next version 😁

shivwolf, 2018-01-23:
McGiverGim

I take this issue to ask for somthing relationated...

Is possible to upload a portable version?
Like 10.0.0
Only to uncompress it and go.

It´s just for use it at work (I have to ask for installation permissions)

Thank you in advance.

McGiverGim, 2018-01-23:
In theory the installer adapts itself to the permissions available. If it does not work, you can copy the installation of your computer and take it off with you. It will work without problem.
The versions available were not a decision of mine 😁

mikeller, 2018-01-23:
@shivwolf: We are trying to minimise the number of different installers we release (and have to support). Can you try to use the installer for a local installation, like @McGiverGim described above, and let us know if you encounter problems somewhere?

closed 2018-05-13 by stale[bot]