betaflight/betaflight-configurator #883: Configurator - Random USB connection sounds.

BadRaven, 2018-01-24:
Since 10.0.0 and even worse with 10.1.0, when Configurator is loaded to either my Vista or my main Windows 7 laptops I am getting random and frequent USB connecting/disconnecting sounds. This only occurs when Configurator is loaded and occurs whether set to automatic or manual port choices.  The sounds also occur as expected on connection or disconnection of a quad or any other USB device.

Configurator appears to work OK, but this occurs whether not doing any action or if in adjustment screens making changes or while firmware flashing.

Rate can be several back to back, then gap of minutes, then once, then gap of 15-30 seconds then another, etc.

The only other USB item in use (on both) are Logitech M185 Mouse. These are unaffected in action, and with or without Configurator loaded work perfectly with no unwanted sounds.

richard-scott, 2018-01-24:
Is your laptop running only on battery power when this happens?

BadRaven, 2018-01-24:
No, both are on mains.

BadRaven, 2018-05-06:
Am I the only one with this? I find it hard to believe!   As soon as Configurator is loaded I can get random amounts of the USB connect/disconnect sounds. I'm up to date on Configurator and it has made no difference. During the average session on Convigurator setting up a new quad, I may get 20 - 50 sounds, sometimes strings of them.
Its not the port
Its not the lead(s)
It ONLY happens when Configurator is loaded.

I do not think it is actually disconnecting and reconnecting the USB lead connected quad.  The only other USB item in use is a Logitech M185 Mouse. (Normally this is quite and faultless, and does not seem to be affected during these sounds.

mikeller, 2018-05-06:
Neither Windows Vista nor 7 are still supported by Microsoft, and official support for them has been dropped by Chrome and by proxy the NW.js toolkit that is used by the configurator. There is nothing we can do to fix this.

closed 2018-05-06 by mikeller

BadRaven, 2018-05-07:
Sorry but that is Incorrect, while improvement support ended (it would, its a stable product), Windows 7 security support won't end until January 14, 2020.

As a stable product (and a lot better than 8.0/8.1) its a more than viable OS and I suspect still widely used.

mikeller, 2018-05-07:
@BadRaven: This is not a security problem, this is simply a matter of current versions of NW.js not supporting Windows 7 any more, which is understandable since Microsoft will only support this operating system for customers on paid support plans. As I said, there is nothing that we can do about this, feel free to open an issue for this against NW.js.

Alternatively, you should still be able to use the Chrome web store version of the application (https://chrome.google.com/webstore/detail/betaflight-configurator/kdaghagfopacdngbohiknlhcocjccjao) if you install it on a version of Chrome that runs on Windows 7.