betaflight/betaflight-configurator #886: Not shown Korean language translation on MAC OS

fixmania, 2018-01-25:
I mention this issue instead of Korean users using MAC OS.

I already have checked the configurator app for MAC OS that includes the 'messeages.json' file translated into Korean language. (located to '_locales\ko folder)

But Korean users using MAC OS said that the app was not shown in Korean language but English.

Also they have set the system language and first order language with Korean.

On the program list in MAC PC, just showed only 1 list(Betaflight Configurator.app). (They has 2 lists -
 Betaflight Configurator, Betaflight Configurator (English)) - on the Windows system.)

I don't know why it happend and can't test any more because I don't use MAC PC.
But Korean users using MAC OS mentioned this issue to me.

Please check it and hope it to be solved.

Added the below capture images.
[image]
[image]

McGiverGim, 2018-01-25:
Hi! The MAC version only has one "shortcut" because we cannot decide the locale to use when starting the app (so we don't have created the "english" shortcut, this is normal in your installation). In theory it will use the korean by default if:
```
To change the locale on Mac, you use the system preferences.

From the Apple menu, choose System Preferences
Under the Personal section, choose International
Choose your language and location
Restart Chrome
```

I don't have a MAC neither, maybe other user can say what happens, but I've code ready to let the user choose the language in a menu in-app, so in the next version this problem will be gone.

fixmania, 2018-01-25:
@McGiverGim I have heard it from some users on online community, also I don't have MAC neither. I will give your answer to them.

Thank you for adding the new menu that can choose any language in app. It will be very useful for all. ^^;

McGiverGim, 2018-05-05:
I close this, now that the language can be chosen from the menu.

closed 2018-05-05 by McGiverGim