betaflight/betaflight-configurator #890: OSD issue: PAL cossshairs not center

Gorilla-1983, 2018-01-26:
The crosshairs for a PAL camera are not in the centre of the OSD, in the Configurator or in the goggles. The crosshairs remain in the centre of what was the NTSC display, according to the Configurator preview.
I have tried to adjust the crosshairs in the OSD tab but they are in a fixed position (unmovable) and in the CLI using the command "set osd_crosshairs = 2318" but they still remain in the same off centre location.
Is there a command I am missing?

McGiverGim, 2018-01-26:
If I'm not wrong, the number of columns is pair, so it is not possible to find a column in the center: or we change the cross to use two columns, or it will always be in the left/right of the center.

Gorilla-1983, 2018-01-26:
I didn't describe well enough when I said off centre; the crosshairs are in the centre left to right but not up and down. They sit roughly 2/5ths from the top of the screen

DanNixon, 2018-01-26:
There is planned work to make the position of the crosshairs configurable.

Gorilla-1983, 2018-01-26:
Brilliant, thanks for getting back to me so quick.

Gorilla-1983, 2018-03-01:
G'day, DanNixon said in a previous post that the crosshair position is to be made configurable. Does any one know if this is available yet?

McGiverGim, 2018-03-01:
It has been planned for 3.4 https://github.com/betaflight/betaflight/pull/5326

Gorilla-1983, 2018-03-01:
Brilliant 😀, thanks for responding so quickly

closed 2018-05-13 by stale[bot]

mmakam2, 2018-06-02:
hi there, I am experiencing this issue in version 3.4 DALRCF405. Any plans for a fix?

mmakam2, 2018-06-02:
[image]

mikeller, 2018-06-03:
@mmakam2: It is fixed. Adjust `osd_crosshairs_pos` to make your crosshairs centered. The (soon to be released) configurator 10.3.0 will allow you to do this in the GUI.

mmakam2, 2018-06-03:
thank you sir