betaflight/betaflight-configurator #908: Entry point not found..

Fastquad, 2018-02-03:
Download Betaflight file, open and install it with no problem.
click the Betaflight icon to open the program..!! 
and little window open up and say  " entry point Release SRW lock Exclusive could not be located in the dynamic link library. KERNELL32.dll."
That is it..!! nothing no Betaflight..!!!
running Win XP 32Bit system.

McGiverGim, 2018-02-03:
The nw.js library used by the configurator don't support windows xp ☹️

McGiverGim, 2018-02-03:
More info here https://github.com/betaflight/betaflight-configurator/issues/819

McGiverGim, 2018-02-03:
Is nothing that we can support or not. The configurator is build over chrome, so we are limited by this tool. Sorry.
You can build the code for chrome and load it as extension manually. By the moment I think it will work.

Fastquad, 2018-02-04:
Hey..!! 
ok, thank you for the info..!! maybe it will,
anyhow,   I think was a bad idea to go with chrome configurator,  big company..!!  don't like big co.!  and here we are..~~~ 
i wonder if can be done as a self support program..!!  change all the codec..!!  BIG...> job for sure.!! 
Well, got my little notebook running on 7, maybe get this one fire-up again and load it there, not sure if i got this running on 32 or 64, have to check it, forgot.
will be only way to get my quad running again soon and change some parameters ..! 
thank you again my friend..!!!

McGiverGim, 2018-02-04:
If your notebook has windows 7 it will work. 32 or 64 bit will work both.

closed 2018-05-13 by stale[bot]