betaflight/betaflight-configurator #909: Not updating

JParkhill1, 2018-02-04:
I have 10.0.0 and trying to get 10.0.1 but its not doing anything when i click the link.

McGiverGim, 2018-02-04:
What link are you referring to?

JParkhill1, 2018-02-04:
When I open BF, it prompts me to update so sends me to GitHub where I click
on the win32 10.0.1 download link. Save As opens up where I go to save but
nothing happens. My CPU is Win64 so maybe that has something to do with it?
I didn't see a link or option for Win64..

McGiverGim, 2018-02-04:
No, It has nothing to do with amd64.
This is the link: https://github.com/betaflight/betaflight-configurator/releases
Did it opens inside the configurator or on a external browser window?

JParkhill1, 2018-02-04:
Weird, that link work It and it downloaded.. thx alot

betaflight-configurator/releases

closed 2018-02-10 by mikeller