betaflight/betaflight-configurator #912: [Question] Configurator 10.2 - {New} 6 rateprofiles - how do I address/select rate profiles 4-6?

chime13, 2018-02-07:
Noticed that 10.2 now has 6 rate profiles as opposed to 3. I can't seem to figure out how to select the last 3 rate profiles.
Thanks

McGiverGim, 2018-02-07:
It depends on the firmware, you need latest version of betaflight 3.3 to see them.

chime13, 2018-02-07:
I do have BF 3.3.0, and I do see, and can program the 6 individual rate profiles.

My question is how to access 4 thru 6 on my TX?

I have a 3pos AUX to address 1 thru 3. I can't seem to find a way to assign the other 3 to another 3pos AUX.

McGiverGim, 2018-02-08:
Maybe this is more a problem of the LUA script. I'm sorry but I've a simple FlySky TX ;)
I think you must add the issue here: https://github.com/betaflight/betaflight-tx-lua-scripts/issues
What do you think? Are you using the LUA scripts?

mikeller, 2018-02-08:
@chime13: More of a question for https://github.com/betaflight/betaflight/issues. Please close here and open there, so others can find it.

richard-scott, 2018-02-08:
Will rate profiles 1 - 3 still be selectable on a 3 position switch using the Adjustments feature on an AUX channel? I don't have LUA capability on my TX.

chime13, 2018-02-09:
@richard-scott  AFAIK, the original 3 rate profiles remain unchanged as far as accessing them from your 3 pos switch. My question was on how to address the last 3....HTH

closed 2018-02-09 by mikeller