betaflight/betaflight-configurator #915: OSX 10.7.5 CANT RUN NEW STAND ALONE APP

spacebubi, 2018-02-09:
i know my mac was oldie....but until this release of new apps, i cant setting up any FC in my house...i cant buy another mac for this thing! please make a compiled version for old mac users (i have osx 10.7.5 and i cant upgrade to newers)

I Need to use my stuff!! now i'm blocked...old version from chrome not working offline..online....nothing ...only CLI appear and i m no sure that working proprly

AndersHoglund, 2018-02-09:
You should still be able to use the latest version by doing a local file load in Chrome.

etracer65, 2018-02-09:
@AndersHoglund I tried downloading the current source and adding as an unpacked extension in Chrome and get errors.

```
Failed to load extension from: ~/development/etracer65/betaflight-configurator
Could not load background script 'js/eventPage.js'.
```
Seems like all of the files are relative to the `src` directory and the manifest lists them relative to the root.  Am I missing something?

McGiverGim, 2018-02-09:
You must pack the extension first. Read the instructions and do a 'gulp release --chromeos'

etracer65, 2018-02-09:
Well that's going to be beyond the skills of most people who are stuck running older versions of OS X.  Perhaps it would make sense to include the Chrome extension version in the releases (at least until Chrome eventually refuses to run them).

McGiverGim, 2018-02-09:
You can use this http://andwho.sytes.net:8080/job/BorisB_BetaFlight_Configurator/ the chromeos version is with the Linux versions 😁

etracer65, 2018-02-09:
Downloaded the Jenkins build and it does register as an extension in Chrome but won't open on a machine with an old OS X version (10.8.6 as a test) which has Chrome 49.  No errors, just does nothing.  Tested the same on an up-to-date machine with current Chrome and it worked.

So I'm guessing it's a Chrome version compatibility issue?  But again back to the same problem as current Chrome only supports OS X 10.9+.  Not really an issue for me but looking for a solution for the original issue opener.

AndersHoglund, 2018-02-09:
I get a manifest error when drag&dropping the BFC chromeos.zip file from Jenkins. 
But when using "Load Unpacked Extensions" from the "dist" subdirectory it all works fine. Provided I have made the build steps (npm install, gulp release). So there is probably only something missing in packing.
This is on latest MacOS and Chrome so it might not be relevant for the original issue, but maybe rather a separate issue on its own.

McGiverGim, 2018-02-09:
The zip includes all the files from the dist folder. Did it work the drag and drop before?

AndersHoglund, 2018-02-09:
I have not tested d&d before, but I think @DieHertz said it used to work.

DieHertz, 2018-02-09:
Nah, I was speaking about installing a packed extension via drag&drop.
Standalone won't work because Chromium won't work on anything less than 10.9.

It should still be possible to install via `Load unpacked extension` and sounds like an user error from what is written here.

etracer65, 2018-02-09:
Loading the current Jenkins build as an unpacked extension worked on a current Chrome version but not in Chrome 49 (the last version supporting earlier than 10.9).  Just does nothing when trying to open.  No errors are reported.

DieHertz, 2018-02-09:
Maybe there's a min version flag in manifest or something?

AndersHoglund, 2018-02-09:
So, users of old Macs with old OS and old Chrome are fucked? No way to run BFC either as a Chrome app (in what ever way it is installed) nor as a standalone application.?  That is a shame... But still, we do not support Win3.11, Win XP etc either.  Just curious, how old are those osx10.7 things anyway?

etracer65, 2018-02-09:
10.7.5 was released in October 2012.

DieHertz, 2018-02-09:
The question is how old is hardware which is limited to 10.7.5.
My Early 2013 MacBook Pro is at High Sierra now, so the Mac in question is probably earlier than 2010

spacebubi, 2018-02-09:
for now, i resolved using on same Os a Virtual machine on VMfusion with
Windows 7, runs extremly slowly but works.
About my issue  I've tryed the hint suggested by Andrey but in all hints no
response from the app (on osx 10.7.5 ).
Thanks to all for patiance and supporting

etracer65, 2018-02-09:
> Maybe there's a min version flag in manifest or something?

The minimum version in the manifest is 38.  Doesn't open with Chrome 49.  Works fine with latest version 63.

AndersHoglund, 2018-02-09:
OK, 18 year old stuff, or more. Should anyone expect that to still be supported? I am in favour of LTS, but seriously...?

etracer65, 2018-02-09:
I'm pretty sure 2018 - 2010 is 8, not 18...  just saying...

mikeller, 2018-02-10:
Yeah, there are limits to what we can support. And since we are relying on Chrome / chromium as the framework to deliver the configurator, and Chrome / chromium moves on, the 'it worked once, so it should work now' argument does not hold either.

A couple of points:
- if the packed Chrome web app can be run on legacy OS versions that otherwise can't run the configurator, then we should start releasing this format;
- at the same time, if this is true, then users of these legacy versions can just install and use the version from the Chrome web store for now. If Chrome will continue to be able to run chrome web apps after Chrome web store support for non - Chromebooks has been removed remains to be seen.

etracer65, 2018-02-10:
The current ChromeOS version from the latest Jenkins build will work as an unpacked extension on current Chrome but won't open on older versions of chrome on old OS X.  The extension will register but does noting when you try to open it.

I found the problem - not sure if we can make a release that supports both current and old Chrome versions.

When opening on an old Chrome version (version 49 in this case), the following error is logged:

```
Error in event handler for app.runtime.onLaunched: Error: Invalid value for argument 2. Property 'icon': Unexpected property.
    at startApplication (chrome-extension://cikiaakhbogolbpemghjblkmahdmkmlo/js/eventPage.js:11:23)
```

I found that removing the `icon` property from `eventPage.js` solves the problem and the app seems to open and run properly.

So this:

```
function startApplication() {
    var applicationStartTime = new Date().getTime();

    chrome.app.window.create('main.html', {
        id: 'main-window',
        frame: 'chrome',
        innerBounds: {
            minWidth: 1024,
            minHeight: 550
        },
        icon: 'images/bf_icon_128.png'
    }, function (createdWindow) {
```

becomes this:

```
function startApplication() {
    var applicationStartTime = new Date().getTime();

    chrome.app.window.create('main.html', {
        id: 'main-window',
        frame: 'chrome',
        innerBounds: {
            minWidth: 1024,
            minHeight: 550
        }
    }, function (createdWindow) {
```

I'm guessing that the `icon` property was added in later versions of Chrome.

DieHertz, 2018-02-10:
Good find, it should be possible to include this field at run time based on chrome version. We should support them if it's that easy and while it still works

spacebubi, 2018-02-11:
{ "manifest_version": 2, "minimum_chrome_version": "38", "version":
"10.2.0", "author": "Betaflight Squad", "name": "Betaflight -
Configurator", "short_name": "Betaflight", "description": "Crossplatform
configuration tool for Betaflight flight control system",
"offline_enabled": true, "default_locale": "en", "app": { "background": {
"scripts": ["js/eventPage.js"], "persistent": false } }, "permissions": [ "
https://maps.googleapis.com/*", "https://*.github.com/", "https://*.
githubusercontent.com/", "http://*.baseflight.net/", "https://*.
amazonaws.com/", "serial", "usb", "storage", "fileSystem",
"fileSystem.write", "fileSystem.retainEntries", "notifications",
"alwaysOnTopWindows", {"usbDevices": [ {"vendorId": 1155, "productId":
57105} ]}, "webview" ], "sockets": { "tcp": { "connect": "*:*" } },
"webview": { "partitions": [ { "name": "map", "accessible_resources" :
["tabs/map.html", "tabs/map.js", "/images/icons/cf_icon_position.png"] } ]
}, "icons": { "128": "images/bf_icon_128.png" } }
here the error when i try to run like you suggest before ETracer65 ,
erasing line about .png

spacebubi, 2018-02-11:
Google Chrome49.0.2623.112 (Build ufficiale) (a 64 bit)
Revisione4fc366553993dd1524b47a280fed49d8ec28421e-refs/branch-heads/2623@
{#663}
Sistema operativoMac OS X Blink537.36
(@4fc366553993dd1524b47a280fed49d8ec28421e)
JavaScriptV8 4.9.385.35Flash21.0.0.213
User-agentMozilla/5.0 (Macintosh; Intel Mac OS X 10_7_5)
AppleWebKit/537.36 (KHTML, like Gecko)
Chrome/49.0.2623.112
Safari/537.36

2018-02-11 18:10 GMT+01:00 Spacebubi <samuele.filippi@gmail.com>:

> { "manifest_version": 2, "minimum_chrome_version": "38", "version":
> "10.2.0", "author": "Betaflight Squad", "name": "Betaflight -
> Configurator", "short_name": "Betaflight", "description": "Crossplatform
> configuration tool for Betaflight flight control system",
> "offline_enabled": true, "default_locale": "en", "app": { "background": {
> "scripts": ["js/eventPage.js"], "persistent": false } }, "permissions": [ "
> https://maps.googleapis.com/*", "https://*.github.com/", "https://*.
> githubusercontent.com/", "http://*.baseflight.net/", "https://*.
> amazonaws.com/", "serial", "usb", "storage", "fileSystem",
> "fileSystem.write", "fileSystem.retainEntries", "notifications",
> "alwaysOnTopWindows", {"usbDevices": [ {"vendorId": 1155, "productId":
> 57105} ]}, "webview" ], "sockets": { "tcp": { "connect": "*:*" } },
> "webview": { "partitions": [ { "name": "map", "accessible_resources" :
> ["tabs/map.html", "tabs/map.js", "/images/icons/cf_icon_position.png"] }
> ] }, "icons": { "128": "images/bf_icon_128.png" } }
> here the error when i try to run like you suggest before ETracer65 ,
> erasing line about .png
>
>
>> Good find, it should be possible to include this field at run time based
>> on chrome version. We should support them if it's that easy and while it
>> still works
>>
>>
>
>
>
> Samuele Filippi

etracer65, 2018-02-13:
By the way, removing the icon property also works to get the Blackbox Viewer running in old Chrome versions (in background.js)

closed 2018-05-13 by stale[bot]

riquezjp, 2018-05-17:
Wondering if there is a way I can hack this to work? I have 2 laptops but both are limited to Chrome 49.

Removing the icon property from eventPage.js creates another error ...

`Could not load background script 'js/eventPage.js' `& shows manifest.json

mikeller, 2018-05-17:
@riquezjp: Try pulling the latest version from the repository, and build the ChromeOS version. The icon issue has been fixed in #985. If this still does not work, please report back.