betaflight/betaflight-configurator #917: Feature Request: Use previously downloaded firmware images.

richard-scott, 2018-02-11:
To save bandwidth is it possible for the configurator to cache previously downloaded firmware hex files so that when/if you need to re-flash the same image you don't have to re-download it?

mikeller, 2018-02-11:
Good idea! We could probably use checksumming to make sure the firmware image is valid.

atomgomba, 2018-03-16:
Amazon is already sending an `ETag` header in the response, so I guess this could be implemented by persisting the file and sending the proper header with the download request.

atomgomba, 2018-04-16:
I'm wondering about cache expiration policy from a usability point of view. Should it be based on time (max. N days) or size (max. N megabytes) of the cache directory? Or maybe put a "Clear download cache" button on the downloader screen and leave it to the user? Or just keep everything and leave it to the user to remove the cache folder manually? Personally I'm leaning towards the "Clear cache button" solution instead of trying to decide what the user wants.

mikeller, 2018-04-16:
Yeah, I would go for manual cache clearing, if at all. After all, the number of firmware files is limited by the number of releases, and the number of board that a user is going to flash, so it's not going to be excessively large (all the firmware files of one release combined are less than 100 MB).

HThuren, 2023-07-17:
@haslinghuis maybe this can be marked as solved with the new way of getting firmware from cloud

haslinghuis, 2023-07-17:
Superseded by: https://github.com/betaflight/betaflight-configurator/pull/3039

closed 2023-07-17 by haslinghuis