betaflight/betaflight-configurator #918: Cant save OSD settings

PickleRick9608, 2018-02-11:
I am using Betaflight configurator 10.1.0 and I cant save my OSD setting on the flight controller.
I updated the Flight controller's firmware to the latest 3.2.5.
The log file on the top says I saved the setting, but in fact none of the OSD positions have been changed, I cant change the Icons postitions or anthing, but the OSD works.

Any idea what sould I do?

richard-scott, 2018-02-12:
I've just had a similar thing. I had to reflash my firmware and it was ok after that.
Also, did you run "defaults" and "save" in the CLI after flashing but before configuration?
That can also help.

mikeller, 2018-02-12:
Good point, @richard-scott. A corrupted configuration is one of the few possible explanations for this, and a reset to defaults / restore will fix that.

PickleRick9608, 2018-02-13:
I changed many things at the same time, so I am nut sure what helped me of them, but it works well after long ours.

I refreshed the firmware of the OSD via FTDI by downloading the MW_OSD firmware/bootloader on it. I set it to communicate with betaflight.
I switch a new serial port to flight controlelr in betaflight and then everything worked well.

I can move the icons on the OSD and save it, and set the PID via my OSD with my transmitter, so am happy :)

Thaks the idea!

mikeller, 2018-02-13:
@PickleRick9608: If this is solved for you, can you close the ticket please?

closed 2018-02-13 by PickleRick9608