betaflight/betaflight-configurator #932: Separation of Roll and Pitch RC-Rate

Macfly581, 2018-02-16:
For people which are not using Mode 2 on here Radio it might be better to change RC-Rate on Pitch and Roll in different ways. For me for example who is using Mode 4, i want the RC-Rate of Pitch an Yaw identical but different RC-Rate on Roll. I am new here so I am not sure is this an Feature request for the configurator or the betaflight code.

wind0r, 2018-02-16:
Betaflight 3.3 allready allows Roll and Pitch with different Rates.
Now the Configurator just needs to add those. The Issue totaly is in the right place here ;)

richard-scott, 2018-03-05:
It would be great when this is implemented if it could also separate the 'RC Expo' values for Pitch and Roll too :-)

closed 2018-05-13 by stale[bot]