betaflight/betaflight-configurator #934: Failsafe works with IBUS - but no parachute icon

stefan73, 2018-02-19:
This might be a bug in the configurator or betaflight. Failsafe works but parachute icon does not.

Setup: 
- Flysky IBUS Receivers; tested the following 3: ia6b, Banggood clone 8A Ver 2.0, X6B
- Flysky Remote FS i6
- F3 & F4 Flight controllers
- connection RX to FC by IBUS

The good news first: Failsafe works reliably. Switch off remote & failsafe procedure starts and motors stop.
The questionmark: The parachute icon in the confugurator 10.1 (tried 10.0 as well ) remains off. This is independent whether I configure a failsafe channel behaviour (throttle at -120%) or not. It seem that all three receivers message the lost connection to the remote over the IBUS protocol. Whether I create an invalid throttle value or not, does not change the behaviour of the quad nor the ICON.

Betaflight also gives reason 3 (RX lost) for disarming on the 1st tab. But as said: No Parachute Icon. In comparison INAV shows the parachute in the configurator.

mikeller, 2018-02-19:
@stefan73: Yes, there seems to be a bug.

> Betaflight also gives reason 3 (RX lost) for disarming on the 1st tab.

This is not a reason for disarming, this is a reason to disable re-arming. Once armed, these 'arming disable' reasons are not checked any more.

L4ky, 2018-03-31:
Same for me. Failsafe stage 1 doesn't turn on Parachute icon. Stage 2 does.

I want to clarify that also for me Failsafe works, it's just a icon issue.

Setup:
9XR Pro with XJT Module
Frsky XM+ Receiver via S.BUS
Target: KakuteF4V2
BF 3.3.0
BF Configurator 10.2.0 Standalone

If you need any information, ask

danybd, 2021-03-24:
Tried for 10.8.0 version same behaviour observed. If have no receiver at all should this be recognized as failsafe? When I turn off my transmitter the channel jump to some kind of default values but no parachute gets active. How does betaflight know transmitter is off?

haslinghuis, 2021-03-25:
Never had a failsafe (using Crossfire) so could not tell about the parachute icon. :-)
But if turning transmitter off motors will stop on the bench and get the `RXLOSS` warning in OSD.

Please take a look at our documentation and wiki:
https://github.com/betaflight/betaflight/blob/master/docs/Failsafe.md
https://github.com/betaflight/betaflight/wiki/Failsafe-(Flight-Controller)

danybd, 2021-03-25:
Thank you @haslinghuis I´ll have a look into the documentation and try to fix this issue (if needed)

kolins-cz, 2022-11-29:
I have the same issue with SBUS receiver. 
[image]

haslinghuis, 2023-06-25:
Failsafe icon represents the failsafe mode induced in mode tab by aux switch

closed 2023-06-25 by haslinghuis