betaflight/betaflight-configurator #936: Gyro and PID frequency drop-downs incorrect if gyro in 1KHz mode

etracer65, 2018-02-20:
Setting gyro_lpf to any value other than "OFF" (0) or "EXPERIMENTAL" (7) puts the gyro in 1KHz refresh.  The gyro and PID frequency drop-downs on the configuration tab incorrectly calculate the various options based on expecting the gyro refresh to be 8KHz.

mikeller, 2018-08-23:
@smoriarty21: Sure, go for it.

etracer65, 2018-08-23:
Sure, but it's a little "complicated" at the moment.  At a high level you just need to check `gyroConfig()->gyro_hardware_lpf` which is passed via MSP to determine that we're in 1KHz sampling mode.  If so, then the `gyro_sync_denom` is always 1 which corresponds to 1KHz.  So the "Gyro update frequency" drop-down only has a single 1 KHz entry.  The "PID loop frequency" scales off of the 1KHz - so `pid_process_denom = 2` mean 500hz, and so on.

Where it gets complicated is that the 1KHz sampling value for gyro_hardware_lpf is 2 for API versions <= 1.40 and will become 1 from a change I just submitted in API version > 1.41 (https://github.com/betaflight/betaflight/pull/6617)

McGiverGim, 2019-04-03:
@smoriarty21 I suppose that you have abandoned this issue. Is that right? Or was fixed some time ago?

haslinghuis, 2026-01-14:
@coderabbitai analyze if this is still an issue