betaflight/betaflight-configurator #938: map.js reference is not correct

JManDoo, 2018-02-22:
When using GPS, the MAP.js is not referenced correctly
I moved the map.js from the /js/tabs/map.js to the /tabs/ folder and this corrected the issue

McGiverGim, 2018-02-22:
Good find. Is better to change the html to reference the right direction of the js file. I can make a PR tomorrow if nobody makes it before.

McGiverGim, 2018-02-22:
@JManDoo if you can, please post you the PR but it's better to change it here:
https://github.com/betaflight/betaflight-configurator/blob/e14f75015a6697471685c1db2143d2423007331b/src/tabs/map.html#L14

JManDoo, 2018-02-22:
I tried that, but it would reference it correctly for me
I edited the main.html to ref it, as well as the html
But the only way it would work was to move it
`<script src="\js\tabs\map.js"></script> `

McGiverGim, 2018-02-22:
The slash must be in the other direction: /

JManDoo, 2018-02-22:
Grr

McGiverGim, 2018-02-22:
You pushed it to the wrong repository 😋

JManDoo, 2018-02-22:
I noticed, I am new to this 😶

JManDoo, 2018-02-22:
is the correct slack link, listed in main.html http://betaflight.tk/slack
When i go there its a squatter site

McGiverGim, 2018-02-22:
No, to register you can use http://betaflight.ch/ after that the chat is here http://betaflightgroup.slack.com

McGiverGim, 2018-02-23:
I had no much time to look into it. You must change here too:
https://github.com/betaflight/betaflight-configurator/blob/d27c10d63ac126b9307d7922d7303af5bc1434e3/manifest.json#L48
But I tried and didn't work neither.

JManDoo, 2018-02-23:
https://github.com/betaflight/betaflight-configurator/pull/940

McGiverGim, 2018-02-28:
I think that this issue can be closed. It has been merged and released in the new version ;)

closed 2018-05-13 by stale[bot]