betaflight/betaflight-configurator #943: New Betaflight Configurator not installing/working on Windows 10

ibgarya, 2018-02-24:
I tried to install the new configurator  on a Windows 10 system which appears to have been successful.  When I launch the app, I get a application window that is completely black and the arrow cursor flashes in the app.  I have tried to uninstall and re-install  the app several times without success.

Any assistance would be greatly appreciated.

Gary

McGiverGim, 2018-02-24:
It seems a problem with the graphic driver or something similar. Your computer/graphic card is old? You have the option to update the driver of the graphic card?

ibgarya, 2018-02-24:
Yes, it is an older laptop.  I will check to see it there is a newer video driver. 
Thanks for the suggestion.
On a side question, I have a desktop that I could never get working with the Chrome version.  Does this new app use the same USB drivers?

ibgarya, 2018-02-24:
I found a newer driver, but that did not change the issue.  Just wanted you to know.  Thanks for the suggestion.

McGiverGim, 2018-02-24:
Yes, the driver is the same. Did you try with the impulse rc driver fixer? https://impulserc.com/pages/downloads

richard-scott, 2018-02-24:
Try a nightly build of the Configurator from here:
http://andwho.sytes.net:8080/job/BorisB_BetaFlight_Configurator/Platform=windows/

ibgarya, 2018-02-24:
Hi McGiverGim - Yes, I had tried the impluse rc driver fix.  I was only able to get the DFU driver on this laptop.  Once I got it working on this system I didn't continue trying on my desktop.

ibgarya, 2018-02-24:
Richard-Scott - I'm trying that now.  Thanks.

ibgarya, 2018-02-24:
I have tried 3 of the daily builds without success.  I also tried lowering my screen resolution without success.  What is interesting is if I hover my cursor over the upper right corner, bubbles appear with baud rate and port labels.  Also, the arrow icon changes to a pointing hand.  Just thought that might be important.

McGiverGim, 2018-02-26:
There are some parameters that you can pass to the executable that maybe make it work... try with this: `--disable-gpu` or wit this `--force-cpu-draw` (or with both).

The complete list is here, bug is huge 😋  https://peter.sh/experiments/chromium-command-line-switches/

ibgarya, 2018-02-26:
> chromium-command-line-switches/

I will give this a try tonight when i get home.  Thanks.

ibgarya, 2018-03-05:
Sorry for not getting back sooner, I've been out of town.  I tried the 10.3
build from today and WIndows defender says it has a virus.  Most likely a
false positive.  Just thought you  should know.

Gary

claytonbakerjr, 2018-03-08:
@ibgarya I read back through this and didn't see mention but have you disabled/turned off your AV software and then tried to install BF Configurator?

ibgarya, 2018-03-08:
Hi,

No, I did not disable my anti-virus.  I did not have issues trying to
install earlier versions of the daily build with my anti-virus.  As a
result, I was not expecting to see that version being denied by
anti-virus.  I am not eager to allow code to run that anti-virus indicates
is infected.

Regards,

Gary

McGiverGim, 2018-03-09:
This has happened with earlier versions too. I've checked in 65 antivirus and and only 2 of them detect something (false positives): https://www.virustotal.com/es/file/cbefc12ab2d2c8f11911e01c45fc6025f55aa33218ff0186ad0f24d4e2d36062/analysis/1520579211/

You can always build the executable yourself, the code is open so there's clear that it has not any virus or trojan.

ibgarya, 2018-03-09:
> 25f55aa33218ff0186ad0f24d4e2d36062/analysis/1520579211/

Good to know.  I will give the new build a try this weekend.  Thanks for
your followup.

Gary

ibgarya, 2018-03-14:
I tried betaflight-configurator-installer_10.3.0_win64 with the same results.  The screen is black, but if you search around, you see the bubbles for different things like Black Box.  When I clicked on it, it took me to a Black Box page.

ibgarya, 2018-05-06:
I have tried the latest build - as of last Wednesday with the same results.

mikeller, 2018-05-06:
Sorry, I don't think there is much we can do - this does not seem to reproduce anywhere but on your machine. Maybe open an issue with NW.js, or check if they already have one open for your particular hardware configuration?

closed 2018-05-06 by mikeller

ibgarya, 2018-05-07:
Thanks for trying.  I do appreciate the effort.

Regards,

Gary

starline, 2022-01-16:
I have the same problem. It arose spontaneously. So far, it helps to run in compatibility mode with win7  sp3

starline, 2022-01-16:
Использование --disable-gpu решило  проблему