betaflight/betaflight-configurator #944: dterm_setpoint_weight has a max of 254, configurator goes to 255

colorex, 2018-02-24:
The slider in the configurator PID tab goes from 0 to 2.55. The max allowed by BF code is apparently 254. When entering 2.55, saving and refreshing (BF 3.3.0 RC2) the value goes to 2.54.

mikeller, 2018-02-24:
@colorex: Can you be a bit more specific about which of the sliders does have the wrong range? An annotated screenshot would help.

EGParadox, 2018-02-25:
[image]
 So it can bee replicated if scroll to 2.55 value hit save>refresh and you get 2.54

colorex, 2018-02-25:
Looks like there was enough information in the OP :)

Yes that’s how you can replicate it. It’s a minor bug without bad
consequences but should be an easy fix anyway.

mikeller, 2018-02-25:
@colorex: Hehe, my bad, reading the text of an issue **and** the title helps.

colorex, 2018-02-25:
I’m new to github. Do I close this issue now that there are commits in place? Or when it’s released? Or will an admin close it?

mikeller, 2018-02-25:
@colorex: Yes please. And thanks for reporting this.

closed 2018-02-26 by colorex