betaflight/betaflight-configurator #958: Missing entries in German locale messages.json

socialmedialabs, 2018-02-28:
Just like the title says. Not a major issue but instead of the window showing "Betaflight Configurator" it just shows "windowTitle".

I added the following to
_/Betaflight Configurator.app/Contents/Resources/app.nw/\_locales/de/messages.jsonmessages.json_
right after the opening bracket:

```
	"translation_version": {
		"message": "0"
	},
	"windowTitle": {
		"message": "Betaflight Configurator",
		"description": "Titel des Anwendungsfensters. Üblicherweise nicht übersetzt."
	},
```
This solves the issue.

McGiverGim, 2018-02-28:
Strange, in theory when the string is missing, it must take the English version, that is the correct form here (the window title must not be translated). I will take a look to it tomorrow.

socialmedialabs, 2018-02-28:
Oh, my bad. Fallback failed to work because I had renamed directories `de` to `en` in order to get German language to work in 10.1.0.
And since it didn't work out of the box I did the same in the current release 10.2.0. However I missed the fact that you now are able to set the language manually, which I did. Hence it failed to reference the original (English) string, because it was in fact looking in the German version.
Hope that makes sense 🙈

closed 2018-02-28 by socialmedialabs

McGiverGim, 2018-02-28:
Ok, understood! Thanks!