betaflight/betaflight-configurator #959: Cannot manually install unpacked extension on Chrome for BFC 10.2.0

WiKDMoNKY, 2018-02-28:
I followed the directions of the BFC GH page like I did for 10.1.0, but I am getting an error.

fujin, 2018-03-01:
@WiKDMoNKY If you can give this nightly build a shot, that would be dope: http://andwho.sytes.net:8080/job/BorisB_BetaFlight_Configurator/Platform=linux/lastSuccessfulBuild/artifact/release/betaflight-configurator_10.3.0_chromeos.zip

WiKDMoNKY, 2018-03-01:
Installing now =)

WiKDMoNKY, 2018-03-01:
@fujin It installs and seems to work. I will play with it more tomorrow. Thanks a million! Any chance you would share how you made it, so I can become self sufficient?

fujin, 2018-03-01:
@WiKDMoNKY In this particular case I just sent you a link to the nightly build (via Continuous Integration) for ChromeOS. If you go [here](http://andwho.sytes.net:8080/job/BorisB_BetaFlight_Configurator/) you will see them all.

In order to build them yourself between nightly/stable releases, you can follow the process outlined here: https://github.com/betaflight/betaflight-configurator#native-app-build-via-nwjs

Roughly:
1) install nodejs, npm, gulp: https://nodejs.org/en/, `npm install --global gulp-cli`
2) run the gulp build task for your platform: `gulp release --chromeos`

Do you think we can close this issue now?

WiKDMoNKY, 2018-03-01:
@fujin One last question... Which Linux distro do I need to install node.js on to build them myself? I haven't touched linux in 10 years or more. I wish there was a step by step tutorial on how to install node.js, download the latest nightly or stable release, then go through all the steps to build it. I would gladly write one up, I just need to figure out how to do it first.

fujin, 2018-03-01:
@WiKDMoNKY Ubuntu is probably the easiest to get started with.

mikeller, 2018-03-01:
@WiKDMoNKY: Can I ask what the reason is for you to go through the trouble of manually installing the unpacked extension? If you want to use the Chrome app version of the configurator (as long as Chrome supports this), it is way easier to install it from the Chrome web store: https://github.com/betaflight/betaflight-configurator#via-chrome-web-store (this way, you'll also get automatic updates).

McGiverGim, 2018-03-01:
@WiKDMoNKY One thing more, you can build it in a Windows too, not only Linux, if you prefer.

adrianmiriuta, 2018-03-05:
@all
Is there a chance to run the betaflight-configurator as unpacked Chrome extension
directly from the cloned git repository ?
The last working md5 is (1875804) 
Then the directory structure changed and It does not work any more.

McGiverGim, 2018-03-05:
You must build it `gulp release --chromeos` or download it from one of the CI servers, for example http://andwho.sytes.net:8080/job/BorisB_BetaFlight_Configurator/Platform=linux/lastSuccessfulBuild/artifact/release/

adrianmiriuta, 2018-03-05:
@McGiverGim 
that means no ?

McGiverGim, 2018-03-05:
No, not directly. You must unpack and build it, not only unpack. You can build the `chromeos` version that will let the directory structure as you need, but need to build it.

adrianmiriuta, 2018-03-05:
@McGiverGim 
thx

until now I did not unpack it ... I used directly the git repository (as unpacked extension)

closed 2018-05-13 by stale[bot]