betaflight/betaflight-configurator #968: Somethings Missing?

claytonbakerjr, 2018-03-06:
So last night I flashed version 3.3.0 on my CL Racing F4S.
Main reason was to take advantage of the separate rates.
I am running BF Configurator 10.2.0
After flashing I seem to be missing this option of separate
rates for Roll and Pitch.
So thinking I was confused I posted this up on the RC Groups
thread and received this response "To the right of the PIDs are your RATES and SUPER RATES.
Previously there was one RC rate setting shared by Pitch and Roll, but they had independent super rates.
Now each ROLL and PITCH have their own set of RC RATE and SUPER RATE"
Which is what I expected....
To add I selected to do a full chip erase prior to flashing which
the Configurator did show doing so.
Here is the version info from the CLI:
Betaflight / CLRACINGF4 (CLR4) 3.3.0 Mar  2 2018 / 03:39:18 (177472b4f) MSP API: 1.37
Attached are 2 screen shots.
One BF 3.2.2 and the other BF 3.3.0
Both show the current Configurator version
Was the above explanation wrong or am I right in feeling I am missing something?
Or....am I really confused :)
Any thoughts?

[image]

[image]

claytonbakerjr, 2018-03-06:
Also, I grabbed my other laptop and loaded BF Conf 10.2.0 on it and I see the same results on the 3.3.0 firmware.

claytonbakerjr, 2018-03-06:
Ok, I have flashed my Hobbywing F4 which uses the OMNIBUSF4SD hex.
Full chip erase and I still see nothing in light of separate rate settings for Roll/Pitch?Yaw
Screen shot is attached.

[image]

claytonbakerjr, 2018-03-06:
Lastly, the 1st round of discovering this was on a Windows 8 laptop and this past was on a Windows 10 laptop. In the event that matters?

fujin, 2018-03-06:
I do not think the configurator work has been done yet. You can test with CLI for now.

claytonbakerjr, 2018-03-07:
@fujin Got ya. So could you pass on the CLI commands by chance?
I was going to ask if this was still limited to CLI
Thanks man

fujin, 2018-03-07:
@claytonbakerjr sure thing:

BF mode: `set rates_type = BETAFLIGHT`
RF mode: `set rates_type = RACEFLIGHT`

Common parameters:
```
get roll_rc_rate
get pitch_rc_rate
get yaw_rc_rate
get roll_expo
get pitch_expo
get yaw_expo
get roll_srate
get pitch_srate
get yaw_srate
```

claytonbakerjr, 2018-03-07:
@fujin  Awesome dude thanks!!!

claytonbakerjr, 2018-03-07:
I will go ahead and close this since it seems its not an option in the GUI at the moment.
Thanks :)

claytonbakerjr, 2018-03-11:
Hi, sorry to reopen this but I found something interesting today...
While out flying with a friend I needed to go into the BF OSD and
adjust some rates, upon doing so I discovered that the 3 rates selections
show within the OSD. So thinking it was just me I grabbed my other 3.3 quad
and it too shows all 3 rates. My friend also recently flashed 3.3 and confirmed
that his too showed all 3. Just to make sure we both were not crazy we checked 
our 3.2 quads and they only show the 2 rate selections, Roll/Pitch and Yaw.
I'm sure you guys already knew this but just wanted to toss this out there again.
It would be nice to see this in the Configurator however for now I can deal with
the Roll/Pitch/Yaw separate rates in the OSD.

closed 2018-04-03 by claytonbakerjr