betaflight/betaflight-configurator #980: Auto-Connect Slider doesn't save state when you exit the App

richard-scott, 2018-03-17:
I've noticed that recently (running https://github.com/betaflight/betaflight-configurator/commit/d2c6fa2556321164b8868430518a8eef66f46a49 from master) that the auto-connect slider forgests its state when you quit the app.

The slider is always off each time you start the configurator even if you enabled this the last time you used the app. It used to remember the last setting you set.

[image]

richard-scott, 2018-03-18:
That said I've just noticed with the Auto-Connect slider disabled when the App starts, as you connect an FC to USB the Configurator automatically connects.

However if I go into CLI, do nothing and click on Black Box (or any other tab) the FC reboots, but the Configurator sticks on the "Waiting for data..." swirl and does nothing until I click the disconnect button. and I've just noticed I have to click "Disconnect" twice for it to actually disconnect.

richard-scott, 2018-03-19:
Just noticed that if I type 'exit' in the CLI then the FC reboots nicely and the Configurator reconnects even with the "auto-connect" slider disabled.

atomgomba, 2018-05-04:
@richard-scott When you type `exit` in the CLI the Configurator itself is not restarted, hence the actual state of the switch remains. However, I can also reproduce this issue. Nor the state of "Auto-Connect" switch, nor the state of "Expert mode" switch saved properly.

richard-scott, 2018-05-08:
All I know is that in previous releases the "auto-connect" slider used to either default to being on, or the "on" state was saved as I didn't need to click "connect" hardly ever. Maybe that was a quirk of it running in Chrome?

closed 2018-06-14 by stale[bot]