betaflight/betaflight-configurator #987: Incorporate Velocity and Position.

Neilgaede, 2018-03-23:
It would be nice to add the new Velocity and Altitude variables to the "Sensors" tab at some point. Thanks!

mikeller, 2018-03-24:
Neither of them are actually sensor values, and in particular for velocity, I can't quite see how you would get a useful readout from this while the craft is tethered. Maybe you can describe what you'd like to use these graphs for?

NeilGaede, 2018-03-24:
True, Perhaps a Velocity and/or Altitude value then next to the X,Y,Z values in the tab that shows the quad orientation when you move the board. It might give a general idea how stable the altitude indication is going to be when using the alt-hold function. Just a thought.

mikeller, 2018-03-24:
@NeilGaede: Yeah, having it as part of the model orientation view makes sense I think.

NeilGaede, 2018-03-27:
Thanks for listening...

closed 2018-03-27 by NeilGaede