betaflight/betaflight-configurator #989: Debian package says licence is proprietary when installing on Ubuntu 17.10

infil00p, 2018-03-25:
When opening the deb in the package manager, I get the following:

[image]

mikeller, 2018-03-25:
Are you referring to the 'Proprietary' listed as license?

McGiverGim, 2018-03-25:
I don't know if this has changed, but some time ago this info was taken from the repository from which the package is installed, so I don't know if this has solution:
https://bugs.launchpad.net/ubuntu/+source/software-center/+bug/435183/comments/5

McGiverGim, 2018-03-25:
Our copyright file (https://github.com/betaflight/betaflight-configurator/blob/master/assets/linux/copyright) does not conform the standard: https://www.debian.org/doc/packaging-manuals/copyright-format/1.0/

But I've tried to conform the standard format and the result is the same. I don't know if I did it wrong or not.

closed 2018-05-13 by stale[bot]