betaflight/betaflight-configurator #997: Super Rate - Max Limit? (Degrees per sec)

NinjaSauce1, 2018-04-03:
I have been looking for some info. What is the maximum rotational degress per second we can have set in Betaflight? Currently, I cannot seem to go past 1998 deg/sec. Is this the maximum limit? I swear that on older versions of Betaflight, I could have my rates well over 2000 deg/sec. But now, any increase in my super rate does not increase the deg/sec, - it only lessens the amount of stick deflection needed to get there. Am I missing something? Or is this just a "safe" maximum now?
I was also wondering - if this is indeed the max limit for the GUI, is there a way to increase it in the CLI? 
Im only asking because I seem to be rapidly outgrowing my rates lately and I have evolved into flying with fairly unusual rates. It would be really helpful to me if I could increase my final deg/sec using the super rates because I would like to keep the rest of my rates the same. 
I cant seem to find much info about this.

McGiverGim, 2018-04-03:
I don't know if this is the reason, but some gyros have problems when exceeding 2000 deg/s: https://github.com/betaflight/betaflight/issues/3959

Arakon, 2018-04-05:
Most of the gyros (at least MPU6xxx) can't handle anything beyond 2000 deg/s. It's a hardware limitation.

mikeller, 2018-04-05:
Not sure if the limitation of commanded spin rate to just below the gyro limit is by design. I think it's just coincidence, and a result of how the rates are applied.

NinjaSauce1, 2018-04-06:
Ok. I can see how it could cause hardware problems. At the same time, I imagine that extreme "jerky" motions could cause problems too. If anything, jerky motions would be worse for the gyro, yet we are able to set the rates so we hit 2000 deg/sec with hardly any stick deflection.
Also, if you couple yaw with flips and rolls, you are amplifying the total degrees/sec. So we are definitely hitting more than 2000 deg/sec.
So maybe the limitation is to protect the hardware.

Is it possible to change the limit in the code?

NinjaSauce1, 2018-04-06:
It appears that the MPU6000 can handle well over 2000 degrees per second. 

The current maximum limit is 1998 degrees per second.

Is this limit set on the FC or Betaflight?

I would like to experiment with faster rates than allowed.
There must be a way to change some code to get around the max limit.

I understand the risks.

borisbstyle, 2018-04-06:
@NinjaSauce1
Mpu600 simply can't measure anything above 2000deg/sec. 

Risk:
When you would command your quad to spin 2005 deg/sec and you reach that value it will think it spins only 2000deg/sec and it will keep speeding up as it thinks it is only spinning 2000deg/sec and could eventually lead into uncontrolled spins.

NinjaSauce1, 2018-05-06:
So I flew with some other pilots last week and we discussed rates. I noticed that their quads were flipping much faster than mine. And their max deg/sec were much lower than mine. They did not even believe me that I was at 2000 deg per sec. So I bumped up my motor timing. It helped a little, but I am certain that my quad is not flipping at the maximum of 2000 deg per sec. So I guess my question is: what is actually controlling the rates? Does the flight controller read the real time data and make itself rotate until it reaches 2000 deg per sec? Or is it a set number where the fc thinks that it can make the motors spin to acheive the desired rate? I will try to watch some DVR footage and do some calculations to determine my actual degrees per second.

McGiverGim, 2018-05-06:
The FC measures gyro and accelerates to the desired rate. Blackbox is the only way to know what happens.

closed 2018-06-12 by stale[bot]