betaflight/betaflight-configurator #1004: unable to enter DFU mode

SkillfulHacking, 2018-04-09:
Mateksys F405 board with betaflight 3.2
attempting to update firmware to latest 3.3.1
board successfully enters bootloader mode (blue led goes out) 
using STM32 bootloader drivers shows up correctly, tried running zdiag and installed WinUSB, also tried Impulse RC driver tool
Symptoms:
Will not flash, betaflight configurator does not allow me to select DFU mode to flash board, COM port disappears once in bootmode

SkillfulHacking, 2018-04-10:
I am using the standalone application version 10.2,0

mikeller, 2018-04-11:
Is anything that is sending data to the flight controller on an UART connected and powered up when connected through USB? If so, disconnect it, or power it down, it is possibly causing DFU mode to bind to the UART in question, and stopping to listen on USB.

Teracis, 2018-04-11:
I have had the same issue with both TinyFish FC and an older Motolab board. I can close the standalone configurator and open the chrome app to have it immediately recognise the board in DFU mode and flash. I've been meaning to bring up the issue.
Windows 10, correct drivers for the FC (they work with chrome configurator), using what I thought was the latest version of the configurator. Doesn't matter what version of firmware the FC is on.

If there's any specific tests or debug data required let me know and I can attempt to collect them.

SkillfulHacking, 2018-04-13:
I was able to resolve my issue by closing the app and re-opening it after the device was in DFU mode.  not sure if that is intended behavior, but I am all set the issue can be closed if that's what is expected.  I

mikeller, 2018-04-13:
It's not intended. But weird behaviour around the Windows drivers is not something we can do much about. Closing this.

closed 2018-04-13 by mikeller

k1dodsoon, 2018-04-16:
Pardon my ignorance,  but I have recently gotten headlight not only laptop win 10, and also cflight,,, although I still cannot understand why I am not able to connect to either with any of my racer quads,,,, swift2 swellpro, darkmax, furibee, custom built x frame with I believe f4 controller,  etc,,?? I've been trying to figure out why this is, for over a year, and have had to pay others to do my damn set ups for me, so frustrating, I mean I am far from an expert at fps, and even less of a computer geek, I've been through atleast 5 laptops thinking that I just had a shitty com, but there is no way I'm mad 5 in a row,,, anyways, I'm also having the same issue using my Samsung tab e, plus my Samsung s7 edge....?? I realize im obviously missing a lot of info, but there is not even one video showing, and also explaining how to get the apps running and also how to get your quads to successfully connect., so I'm hoping someone will help a rookie student out... thanks