betaflight/betaflight-configurator #1006: Cannot find FC on MacOS

wx4cb, 2018-04-17:
Hi all

Not strictly a configurator issue par se as i'm not sure if this is a driver issue or what. I have just bought a new macbook pro which has all the latest updates etc from apple.

When i plug my flight controller in, it lights up just as it should but i have no serial port on the computer.

I have downloaded the cp210x drivers and installed them, and im looking for stm32 drivers, but alli can find is windows drivers?

any help would be appreciated.

richard-scott, 2018-04-17:
Did you try something like /dev/tty.SLAB_USBtoUART as the port?
I found [this ](http://blog.sam-thompson.info/cleanflight-on-mac-osx/) on Google.

wx4cb, 2018-04-17:
The Army TTY devices I see that are not the normal ones they Bluetooth on.
C n e t t y s l a b or anything like that I'm being new to Mac and I don't
know how to test to see if it actually sees it you know like a device
manager

wx4cb, 2018-04-17:
Okay I don't know what I did differently but it works now for some reason I guess maybe I had it in the wrong USB 3 Port but I don't think so because the Apple Hub I have full this Mac Pro has two USB C ports for inputs which match up with the two USB C ports that are on the Mac Pro but I just tried it again and it seems to be okay it sees them as USB modems though but it worked

closed 2018-04-17 by wx4cb