betaflight/betaflight-configurator #1008: JETIEXBUS Serial Receiver Provider not working

paulboes73, 2018-04-19:
During set up in the configurator when selecting serial receiver provider as JETIEXBUS the configurator will reboot and set values to null in the receiver section.  When going to the receiver tab, nothing shows up.  I'm using Configurator 10.2.0 and betaflight 3.3.1.

mikeller, 2018-04-20:
@paulboes73: What flight controller are you using?

paulboes73, 2018-04-20:
betaflight f3 flight controller

paulboes73, 2018-04-20:
Tested 3.2.5 build and there is no issue.

mikeller, 2018-04-20:
JETIEXBUS has had to be removed from the standard builds for F3 targets to make room in the flash for other new features that were added, sorry.

If you want to use JETIEXBUS on the BETAFLIGHTF3, you'll have to do a custom build, enabling JETIEXBUS and disabling other RX protocols.

closed 2018-04-20 by mikeller

paulboes73, 2018-04-20:
Will JETIEXBUS work for f4 and f7 boards?

mikeller, 2018-04-20:
@paulboes73: Yes. F4 and F7 have more flash space and do not require the removal of features to make the firmware fit (as of now).