betaflight/betaflight-configurator #1016: Restore dosen't work

LexGhost, 2018-05-05:
Hello,
it seems that the resotre buton dosn't work in my Betaflight. In the log ther is nothing if i try to restore my settings like the configurator do nothing when i try to restore something. Backup work fine. The Backup is created with the same configurator on the same version. I don't know whats wrong. I copy the backup i create in here.

Lux F3 FC with btfl 3.1.7
Betaflight Configurator 10.2.0
[BTFL_backup_Realacc210_20180425_103724 - Kopie.txt](https://github.com/betaflight/betaflight-configurator/files/1976954/BTFL_backup_Realacc210_20180425_103724.-.Kopie.txt)

johmarjac, 2018-06-19:
Additionally there seems to be some values missing. ESC Type didnt get saved on OMNIBUS (didnt try another FC yet) so if I selected DSHOT300, backup that, flash betaflight with full chip erase and restore, it put ONESHOT ... So that didnt restore correctly.
Same with OSD, whole OSD settings are not saved, and in OSD Tab there is Alarm values for altitude and power. Power value (default 2200) doesnt get saved also. ( I dont know for Altitude tho, never tried that).

smoriarty21, 2018-06-23:
Thanks @johmarjac I tested this out and sure enough it's doing the same for me.  If nobody has any objections I'm going to take a crack at squaring this away this weekend

Docteh, 2019-10-28:
Which code? right above your comment I see a reference to PR #1083 which was merged back around 10.4.0 maybe there is more recent issues? Personally I've been using `diff` instead of the backup and restore functions.

tracer99, 2019-10-28:
That would be the only code. Looking at the release notes for 10.6.0, getting backup and restore working was one of the features. It's not working. I didn't open a new ticket because this one was still open. 

Using diff is great, but if you have buttons in your UI for a feature, it should work. Or get rid of them.

mikeller, 2019-10-29:
@tracer99: One of the problems of the Backup / Restore buttons is that they currently require the code behind them to be updated for pretty much every release, which is missed often. They also cannot backup / restore settings that are not available outside of the CLI - currently only a `diff` can do that.
But there are plans to change the way that configurator communicates with the firmware which, when implemented, will remove these limitations. So keeping what we have for now probably makes sense.

Can you please open a new issue for the error you are getting when trying to use the backup / restore buttons, and we can look into it?

haslinghuis, 2021-06-05:
Yes. We need to remove this feature I think. As each version has a different set of parameters

closed 2021-06-05 by haslinghuis