betaflight/betaflight-configurator #1028: DTerm Setpoint Weight inconsistent help tooltop

Cleric-K, 2018-05-18:
Ever since https://github.com/betaflight/betaflight/commit/ff1f9ce77#diff-c5f9576172e273033324ba218524cc97L163 the meaning of the Setpoint Weight number has changed.

Previously the raw byte value of the setpoint weight was divided by 100.0:
`dtermSetpointWeight = pidProfile->dtermSetpointWeight / 100.0f;`
which meant that when value of 1.0 (which is raw byte 100) is set in the configurator that would correspond to `dtermSetpointWeight  = 1.0` thus we have D term from error.

Since the above mentioned commit the raw number is divided by 127:
`dtermSetpointWeight = pidProfile->dtermSetpointWeight / 127.0f;`
Which means if we want D term from error (`dtermSetpointWeight  = 1.0`) we have to set the value in the configurator to 1.27.

The tooltip help still shows the old situation:
> Value of 0 equals to old Measuemenent method where D only tracks gyro, while value of 1 equals to old Error method with equal gyro and stick tracking ratio.

What would be the best approach to fix this?
1. Fix the configurator to multiply the weight value with 127 when converting to byte.
2. Change the tooltip help to say that `... 1.27 equals to the old Error method ...`

McGiverGim, 2018-05-18:
I think is good that the configurator uses the same units as the CLI command, so to me is better to change the message. Is a strange message, but maybe something like `...a value of 50% (1.27) equals to the old error method...` in this way maybe it has some sense this strange value. What do you think?

Cleric-K, 2018-05-18:
I agree completely that it would be very confusing if the configurator makes scaling behind the scenes.
Good idea for the 50% addition!

Just one question: this commit has been included from v3.1.6 onwards. Is there a way to make conditional tooltip based on version?

McGiverGim, 2018-05-18:
Yes, it is possible, but not as easy. If you want make it conditional, you must add the tooltip in the `.js` file with Javascript code. I don't know if the effort for this is necessary. @mikeller what do you think about it? Make it conditional? Put the message for the latest version only? Modify the tooltip to include versions information?

mikeller, 2018-05-18:
In general, we make the configurator version-aware, so the configurator should be showing the right tooltip and using the right scaling to match the firmware version.

closed 2018-06-07 by Cleric-K