betaflight/betaflight-configurator #1046: Add Text Warnings to Configurator for DSHOT Beacon User

EggsBenedict, 2018-06-09:
_As a followup to: https://github.com/betaflight/betaflight/issues/6066_

Would you please consider adding a warning to the configurator addressing some of the issues with the DSHOT beacon that could be placed somewhere in here: 
[image]

The first is that placing DSHOT beacon on the pre-arm switch or leaving it on near arming can interfere with “flip over after crash mode” and prevent it activating or deactivating.  Best case is that turtle mode doesn’t work. **Worst case** is that turtle mode doesn’t work AND can create a runaway take off scenario.

Just a stab at wording for that:
> **WARNING**: DSHOT beacon should not be active near or during arming (e.g. configured to activate on the pre-arm switch) as it can interfere with other features.

Secondly, and this can be ignored if you feel it’s too "surgeon's general warning" for you, but the DSHOT beacon can consume a lot of power.  There have definitely been reports of people smoking motors when using it (probably a combination setting it very loud and using it for too long).  This issue strikes me as one that requires the user to connect _A LOT_ of dots to understand the issue but DSHOT beacon is something that lots of people use so people will likely run into occasional issues for quite some time.

Perhaps an additional warning?
> Prolonged use of DSHOT beacon (especially when configured to be loud) can, in rare cases, cause damaging level of heat build up in motors.

Thanks BF devs!

EggsBenedict, 2018-06-11:
HOT DAMN! You guys are studs. This is great! Thanks so much for taking the time to look at this and make some great changes.

McGiverGim, 2018-07-02:
It seems this issue has been fixed, so I close it. Thanks for the suggestion!!

closed 2018-07-02 by McGiverGim