betaflight/betaflight-configurator #1050: Backup button not working (no pop up)

Redeque, 2018-06-11:
In betaflight config 10.2.0 the downloads buttons are not working.
When clicking there is not pop-up for file download location.
This is the case for 'Backup' and 'Save flash to file..."

Seems to be the same issue as:
https://github.com/betaflight/betaflight-configurator/issues/461

Log result for 'both buttons'

`backup_restore.js:192 User cancelled
(anonymous) @ backup_restore.js:192
(anonymous) @ extensions::fileEntryBindingUtil:55
safeCallbackApply @ extensions::uncaught_exception_handler:27
handleResponse @ extensions::sendRequest:52

onboard_logging.js:447 User cancelled
(anonymous) @ onboard_logging.js:447
(anonymous) @ extensions::fileEntryBindingUtil:55
safeCallbackApply @ extensions::uncaught_exception_handler:27
handleResponse @ extensions::sendRequest:52`

mikeller, 2018-06-12:
What is your craft's name?

See https://github.com/betaflight/betaflight-configurator/issues/461#issuecomment-285255652 for an explanation of what's happening, the `User cancelled` is a giveaway.

Redeque, 2018-06-12:
'3" OmnibusF4'

Well yeah it is but there is not solution here.
Reinstalling has to effect since I also tried.

mikeller, 2018-06-12:
@Redeque: I think there is. `"` isn't supported by Windows in file names, that's why it's failing for you. Just rename your craft to `3in OmnibusF4`, and it should work.

Redeque, 2018-06-12:
Guess it does say it just didn't read correctly.
My apologies and thank you for your help!

Best to maybe add a note in the configurator. 

"Ok. I think I can see what goes wrong. Looks like Chrome auto-closes the file selection dialog if the suggested *file name contains characters that cannot be part of a filename*, which is the case with the currently released version under windows. Unfortunately it does so and logs this with the utterly unhelpful message 'User cancelled'.

I will get a new version released."

closed 2018-06-12 by Redeque