betaflight/betaflight-configurator #1063: Build Type select for firmware flasher intermittent issue

richard-scott, 2018-06-18:
I'm testing with an SP Racing F7 Dual FC and when I try and flash it seems like the "Development" dropdown gets populated with the "Release" list of boards when BFC first starts. The BFC remembers I had selected a "Development" build type last time I ran the app, but it doesn't seem to load the correct list of boards on the nexxt run.

Here is the list when I first open BFC:

[image]

And this is the list after I change "Development" to "Release" and back to "Development".

[image]

As you can see, SPRACINGF7DUAL is not in the first screenshot, but is in the second one.

McGiverGim, 2018-06-19:
@DieHertz can you help with this?

DieHertz, 2018-06-19:
Yes, once I get home

PR #1069 (2018-06-20): Fixed CI build type silently defaulting to Release
Fixes #1063
closed 2018-06-20 by DieHertz