betaflight/betaflight-configurator #1064: Betaflight Configurator disappears after any system upgrade (GNU/Linux Debian)

blackmennewstyle, 2018-06-19:
Everytime i upgrade my GNU/Linux Debian system (aptitude update && aptitude safe-upgrade), the Betaflight Configurator disappears from my list of installed programs or packages.

It does actually the same with the Betaflight blackbox-log-viewer.

`hostnamectl Static hostname: debianmsi Icon name: computer-laptop Chassis: laptop Operating System: Debian GNU/Linux 9 (stretch) Kernel: Linux 4.9.0-4-amd64 Architecture: x86-64`

`lsb_release -da No LSB modules are available. Distributor ID: Debian Description: Debian GNU/Linux 9.4 (stretch) Release: 9.4 Codename: stretch`

binlan, 2018-06-24:
what do you mean with "list of installed packages"?
`dpkg -l|grep -i betaflight`
is listing me both on my system. 
`lsb_release -da
No LSB modules are available.
Distributor ID: Debian
Description:    Debian GNU/Linux testing (buster)
Release:        testing
Codename:       buster`

after each run of save-upgrade the both get uninstalled? if so, it sounds like a dependency issue. please post the output of safe-upgrade.

blackmennewstyle, 2018-06-25:
Hi @binlan :)

I did not write they are uninstalled, i wrote they disappeared from my list of applications/packages.

Normally, in the Gnome interface, the app is located in Applications > Other. They disappears from this location after an upgrade.

I also created shortcuts in "favorites", which is a tab which appears on the left of the screen with all applications marked as "favorites". The shorcuts disappear after an upgrade.
This tab has also a button called "Show Applications" which listed all applications/packages installed and the applications are not listed there as well after an upgrade.

It sounds just like more a desktop application issue with the .deb files of betaflight configurator and blackbox

binlan, 2018-06-26:
if you run this:
`xdg-desktop-menu install /opt/betaflight/betaflight-configurator/betaflight-configurator.desktop`
after an upgrade, is it visible again?
if so, i would think this is a bug in your gnome desktop and you could use the command above to workaround that. 
on my machine i have another desktop environment and i cant reproduce your issue.

blackmennewstyle, 2018-06-26:
Yes that command works after an upgrade, it's the command i execute sometimes in order to make both applications to be visible again. Or i just execute another dpkg -i with the necessary .deb files.

I'm afaraid i won't be the only one with the issue. I installed a second virtual machine yesterday with a Debian 9 Gnome and the issue appears there as well, so definitely a problem with Gnome environment.

I'm still thinking a step is missing in the .deb file of betaflight configurator and blackbox because the other applications; i installed manually with dpkg like Virtual Box, Google Chrome or Opera; don't disappear after an upgrade.

binlan, 2018-06-26:
could you try to link the .desktop file to ~/.local/share/applications/ ?
if that helps, the postinst script of betaflight-configurator could do the same. (but to the system folder /usr/share/applications)

blackmennewstyle, 2018-06-26:
I was about to post this idea of copying the desktop file into ~/.local/share/applications/

I was investigating about desktop application on GNU/Linux Debian and some people suggested this trick.

I just gave a try on my VM and it worked out. I tried to upgrade my system or installed a new package (which can also make the application disappears) and the app doesn't disappear and is still visible.

It worked also with the betaflight blackbox application :+1:

binlan, 2018-06-28:
so, i think its a gnome issue and not a packaging issue of btfc. closing this issue?

blackmennewstyle, 2018-06-28:
Not really convinced by your conclusion LOL Lot of packages are not impacted by this problem.
But you can do what you want to do.

McGiverGim, 2018-06-28:
@blackmennewstyle I'm not a Linux expert. It seems some systems (Debian in your case) can have this problem. Under Ubuntu I think I tested it without problems.

If you are able to detect the changes needed in the post install script, we can change it.

haslinghuis, 2021-06-05:
Closing as inactive for more than a year. I have updated documentation for latest versions for latest implementation except Fedora

closed 2021-06-05 by haslinghuis