betaflight/betaflight-configurator #1073: error can't open configurator

PIRATES3D, 2018-06-23:
error can't open configurator
[image]

thanks in advance for the great work.

samuelrettore, 2018-06-23:
I have same problem

arierep60, 2018-06-23:
I have the same error here :(

mikeller, 2018-06-23:
@PIRATES3D, @samuelrettore, @arierep60: A mention of what operating system and version you are using would be real helpful.

arierep60, 2018-06-23:
Windows 10 Home 64 bits

AndersHoglund, 2018-06-23:
Seems to be a lingo issue. Seen other reports, hinting it could be a Spanish or/and Portuguese xlat problem. My English Win10 installation worked flawlessly. Or maybe it worked because I had earlier versions installed, i.e. update works but new installs fail. Just an idea...

PIRATES3D, 2018-06-23:
@mikeller : thanks for the fast answer. I have windows 10 PRO 64bit (Portuguese language) with all the updates installed.

Error message translated: Error loading the extension. Fail at loading the extension from: C:\Program Filles........

Thanks again.

mikeller, 2018-06-23:
@AndersHoglund: Good catch, could be an illegal placeholder (or a character sequence being interpreted as one) in one or more of the language files.

@PIRATES3D, @samuelrettore, @arierep60: Can you please also post what language you are using? For now, the workaround seems to be to switch to English.

McGiverGim, 2018-06-24:
By the error, it seems this phrase `Invalid image size; expected $1×$2 instead of $3×$4` but I'm out of a computer until Tuesday to debug it.

binlan, 2018-06-24:
duplicates #1074

mikeller, 2018-06-24:
Ok, I think I've fixed it (I cannot reproduce it). Please try the configurator builds from https://drive.google.com/drive/folders/1b48TjU2cXBPWrNFA44rOzS3z--txlsw0?usp=sharing, and provide feedback.

binlan, 2018-06-24:
@mikeller does it mean, the workaround with spaces is not needed anymore? (as mentioned in #1074 )

mikeller, 2018-06-24:
Since the strings in question are not used anywhere: No.

arierep60, 2018-06-24:
@McGiverGim did that and now it works! thanks man ;)

arierep60, 2018-06-24:
@PIRATES3D did that and now it works fine, thanks man ;)

AndersHoglund, 2018-06-24:
Would this fix lead to a 10.3.x_maint branch? And a 10.3.1 release? Or should we stay on the "one-track" policy and fix this in the next 10.4.0 release?

mikeller, 2018-06-24:
@AndersHoglund: Yes, I think this will best be fixed / released as 10.3.1 from a maintenance branch, the same way we are doing patch releases for the firmware.

GeraldoGarcia, 2018-06-25:
I am sure @PIRATES3D is using Win10 Brazilian Portuguese version. I had the same problem with the pre-release version from Boris B page and solved it by altering the system default language to English-US.

closed 2018-06-26 by mikeller