betaflight/betaflight-configurator #1075: GPS altitude shows 10 times too high - Configurator 10.3.0

jsgoldberger, 2018-06-24:
On the GPS tab the altitude shows 10x what it should.

AirBreak69, 2018-06-24:
As discussed in BF PR #6069 https://github.com/betaflight/betaflight/pull/6096 the behaviour will change from x10 to 1/10th.
I think in
https://github.com/betaflight/betaflight-configurator/blob/9461217d13faf1ce034bfeab77fc339f5e26263f/src/js/tabs/gps.js#L51
the by 10 division had to ommited with the next BF release because the latest BF code transmit pure meters again

jsgoldberger, 2018-06-27:
With Configurator 10.3.1 and BF 3.4-RC3 I get the following results: GPS tab in configurator shows 8.7m and OSD shows 284ft, this with BARO disabled so altitude is presumably coming from GPS.  Correct GPS altitude would be 87m.  My iPhone shows altitude of 290ft in the compass app.

mikeller, 2018-06-27:
Ok. I looked into this last night, but it looked like firmware 3.4.0-RC3 and firmware 10.3.0 would work out correct, so I did not change it for 10.3.1. Will change it for the next configurator release. What operating system do you use? (So I can build a configurator version for you to test.)

closed 2018-08-03 by stale[bot]