betaflight/betaflight-configurator #1077: 10.3.0 Pixels not detected correctly with custom OSD logo

HaroldBawls, 2018-06-24:
I attempted making my own OSD logo and the colors are not being detected correctly. 

I have tried using this image and it doesn't work. 

[image]

McGiverGim, 2018-06-28:
This image has some pixels that are a mix between green and white, for example in the "BETA" letters limits, between black and green in the wasp limits, etc.

You must only use pure black, white and green, nothing more.

@atomgomba can you confirm this?

mikeller, 2018-06-28:
I can confirm this. After testing this functionality, and struggling to create an image that is accepted, I think there is room for improvement. I think if we changed the processing to use black (#000000) as black, white (#ffffff) as white, and everything else as transparent it would become easier for users to use, without losing any functionality. What do you think, @atomgomba?

atomgomba, 2018-06-28:
There is a green pixel on the image of color `rgb(0, 227, 0)` (`#00E300`). I think these kind of errors stem in the fact that the used image editor is not suited or properly set up for pixel art. For instance it's doing anti-aliasing, "optimize" the palette or what not. I think I'm going to add the hex code and coordinates of the wrong pixel detected to the error message.

McGiverGim, 2018-07-12:
This was fixed in https://github.com/betaflight/betaflight-configurator/pull/1101 but I left this open because the suggestion of mikeller I think is a good idea. Maybe in a future can be implemented.

haslinghuis, 2024-05-06:
Closing as fixed in #1101 and stale request.

closed 2024-05-06 by haslinghuis