betaflight/betaflight-configurator #1079: Failed to open serial port

realflight7, 2018-06-24:
betaflight-configurator = 10.3.0
OS = Linux mint, 18.3
repeatable = yes

When changing a parameter that requires a FC reboot, the serial port fails to open, the only way to regain access is to close and re-open the configurator

2018-06-24 @ 11:08:41 -- Serial port successfully closed
2018-06-24 @ 11:08:44 -- Serial port successfully opened with ID: 3
2018-06-24 @ 11:08:44 -- MultiWii API version: 1.39.0
2018-06-24 @ 11:08:44 -- Flight controller info, identifier: BTFL, version: 3.4.0
2018-06-24 @ 11:08:44 -- Running firmware released on: Jun 17 2018 08:39:32
2018-06-24 @ 11:08:44 -- Board: MKF4, version: 0
2018-06-24 @ 11:08:44 -- Unique device ID: 0x2000343436511438363536
2018-06-24 @ 11:08:44 -- Craft name:
2018-06-24 @ 11:08:44 -- Arming Disabled
2018-06-24 @ 11:09:13 -- EEPROM saved
2018-06-24 @ 11:09:13 -- Serial port successfully closed
2018-06-24 @ 11:09:16 -- Failed to open serial port
2018-06-24 @ 11:09:20 -- Failed to open serial port

mikeller, 2018-06-24:
@realflight7: Hmm, that's odd. I am using linux too and it's working fine for me. Can you do a `dmesg | tail -50`, `tail -50 /var/log/syslog`, and `tail -50 /var/log/daemon.log` right after reproducing the error, to check if there is something else claiming the port? (Paste another copy of the configurator log as well, as a reference for the timestamps.)

realflight7, 2018-07-14:
Sorry for the delay, here are the outputs, this time all I did was turn on airmode:

2018-07-14 @ 13:40:39 -- Running - OS: Linux, Chrome: 67.0.3396.62, Configurator: 10.3.1
2018-07-14 @ 13:40:40 -- Loaded release information for configurator from GitHub.
2018-07-14 @ 13:40:45 -- Serial port successfully opened with ID: 1
2018-07-14 @ 13:40:45 -- MultiWii API version: 1.39.0
2018-07-14 @ 13:40:45 -- Flight controller info, identifier: BTFL, version: 3.4.0
2018-07-14 @ 13:40:45 -- Running firmware released on: Jun 17 2018 08:39:32
2018-07-14 @ 13:40:45 -- Board: MKF4, version: 0
2018-07-14 @ 13:40:45 -- Unique device ID: 0x2000343436511438363536
2018-07-14 @ 13:40:45 -- Craft name:
2018-07-14 @ 13:40:45 -- Arming Disabled
2018-07-14 @ 13:41:04 -- EEPROM saved
2018-07-14 @ 13:41:05 -- Serial port successfully closed
2018-07-14 @ 13:41:07 -- Failed to open serial port
2018-07-14 @ 13:41:13 -- Failed to open serial port

ian@ian-HP-Notebook ~ $ dmesg | tail -50
[212231.990389] sd 2:0:0:0: [sdb] tag#0 CDB: Write(10) 2a 00 01 bb 56 e8 00 00 f0 00
[212231.990394] print_req_error: I/O error, dev sdb, sector 29054696
[212231.990407] Buffer I/O error on dev sdb1, logical block 3631833, lost async page write
[212231.990422] Buffer I/O error on dev sdb1, logical block 3631834, lost async page write
[212231.990430] Buffer I/O error on dev sdb1, logical block 3631835, lost async page write
[212231.990438] Buffer I/O error on dev sdb1, logical block 3631836, lost async page write
[212231.990446] Buffer I/O error on dev sdb1, logical block 3631837, lost async page write
[212231.990453] Buffer I/O error on dev sdb1, logical block 3631838, lost async page write
[212231.990460] Buffer I/O error on dev sdb1, logical block 3631839, lost async page write
[212231.990467] Buffer I/O error on dev sdb1, logical block 3631840, lost async page write
[212231.990474] Buffer I/O error on dev sdb1, logical block 3631841, lost async page write
[212231.990482] Buffer I/O error on dev sdb1, logical block 3631842, lost async page write
[212232.001348] sd 2:0:0:0: [sdb] tag#0 FAILED Result: hostbyte=DID_NO_CONNECT driverbyte=DRIVER_OK
[212232.001359] sd 2:0:0:0: [sdb] tag#0 CDB: Write(10) 2a 00 01 bb 57 d8 00 00 f0 00
[212232.001365] print_req_error: I/O error, dev sdb, sector 29054936
[212922.475984] vboxdrv: 0000000000000000 VMMR0.r0
[212922.825626] vboxdrv: 0000000000000000 VBoxDDR0.r0
[224944.375229] usb 1-3: new full-speed USB device number 54 using xhci_hcd
[224944.516722] usb 1-3: New USB device found, idVendor=0483, idProduct=5740
[224944.516731] usb 1-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[224944.516736] usb 1-3: Product: MatekF4
[224944.516741] usb 1-3: Manufacturer: Betaflight
[224944.516746] usb 1-3: SerialNumber: 0x8000000
[224944.517832] cdc_acm 1-3:1.0: ttyACM0: USB ACM device
[225002.884103] cdc_acm 1-3:1.1: urb 2 failed submission with -19
[225002.884140] usb 1-3: USB disconnect, device number 54
[225002.884165] cdc_acm 1-3:1.1: urb 3 failed submission with -19
[225002.884199] cdc_acm 1-3:1.1: urb 4 failed submission with -19
[225002.884230] cdc_acm 1-3:1.1: urb 5 failed submission with -19
[225002.884326] cdc_acm 1-3:1.1: urb 6 failed submission with -19
[225002.884371] cdc_acm 1-3:1.1: urb 7 failed submission with -19
[225002.884427] cdc_acm 1-3:1.1: urb 8 failed submission with -19
[225002.884461] cdc_acm 1-3:1.0: acm_ctrl_irq - usb_submit_urb failed: -19
[225002.884568] cdc_acm 1-3:1.1: urb 9 failed submission with -19
[225002.884640] cdc_acm 1-3:1.1: urb 10 failed submission with -19
[225002.884709] cdc_acm 1-3:1.1: urb 11 failed submission with -19
[225002.884752] cdc_acm 1-3:1.1: urb 12 failed submission with -19
[225002.884799] cdc_acm 1-3:1.1: urb 13 failed submission with -19
[225002.884852] cdc_acm 1-3:1.1: urb 14 failed submission with -19
[225002.884921] cdc_acm 1-3:1.1: urb 15 failed submission with -19
[225002.884992] cdc_acm 1-3:1.1: urb 0 failed submission with -19
[225002.885049] cdc_acm 1-3:1.1: urb 1 failed submission with -19
[225002.885278] cdc_acm 1-3:1.0: failed to set dtr/rts
[225005.587350] usb 1-3: new full-speed USB device number 55 using xhci_hcd
[225005.733352] usb 1-3: New USB device found, idVendor=0483, idProduct=5740
[225005.733355] usb 1-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[225005.733357] usb 1-3: Product: MatekF4
[225005.733358] usb 1-3: Manufacturer: Betaflight
[225005.733360] usb 1-3: SerialNumber: 0x8000000
[225005.734002] cdc_acm 1-3:1.0: ttyACM0: USB ACM device

ian@ian-HP-Notebook ~ $ tail -50 /var/log/syslog
[... 28 lines cut ...]
Jul 14 13:41:04 ian-HP-Notebook kernel: [225002.884640] cdc_acm 1-3:1.1: urb 10 failed submission with -19
Jul 14 13:41:04 ian-HP-Notebook kernel: [225002.884709] cdc_acm 1-3:1.1: urb 11 failed submission with -19
Jul 14 13:41:04 ian-HP-Notebook kernel: [225002.884752] cdc_acm 1-3:1.1: urb 12 failed submission with -19
Jul 14 13:41:04 ian-HP-Notebook kernel: [225002.884799] cdc_acm 1-3:1.1: urb 13 failed submission with -19
Jul 14 13:41:04 ian-HP-Notebook kernel: [225002.884852] cdc_acm 1-3:1.1: urb 14 failed submission with -19
Jul 14 13:41:04 ian-HP-Notebook kernel: [225002.884921] cdc_acm 1-3:1.1: urb 15 failed submission with -19
Jul 14 13:41:04 ian-HP-Notebook kernel: [225002.884992] cdc_acm 1-3:1.1: urb 0 failed submission with -19
Jul 14 13:41:04 ian-HP-Notebook kernel: [225002.885049] cdc_acm 1-3:1.1: urb 1 failed submission with -19
[... cut ...]
Jul 14 13:41:24 ian-HP-Notebook ModemManager[930]: <info>  Creating modem with plugin 'Generic' and '1' ports
Jul 14 13:41:24 ian-HP-Notebook ModemManager[930]: <warn>  Could not grab port (tty/ttyACM0): 'Cannot add port 'tty/ttyACM0', unhandled serial type'
Jul 14 13:41:24 ian-HP-Notebook ModemManager[930]: <warn>  Couldn't create modem for device at '/sys/devices/pci0000:00/0000:00:14.0/usb1/1-3': Failed to find primary AT port

ian@ian-HP-Notebook ~ $ tail -50 /var/log/daemon.log
tail: cannot open '/var/log/daemon.log' for reading: No such file or directory

mikeller, 2018-07-14:
@realflight7:

> ...ModemManager...

ModemManager is known to cause problems when using non-modem devices that use the ACM device type. See https://github.com/betaflight/betaflight/wiki/Installing-Betaflight#troubleshooting-installation-problems for how to disable it.

closed 2018-08-21 by stale[bot]