betaflight/betaflight-configurator #1080: Configurator 10.3.0 fails to start on Mac OS 10.9.5

furagu, 2018-06-24:
Hello.

When I try to start the Configurator, I get this message instead 

[image]

I thought it was related to #1073 since my system's language was set to Russian, but switching to English did not fix the problem.

Here's the debugging information I get when pressing "Report..." in that error popup.

```
Process:         nwjs [585]
Path:            /Applications/Betaflight Configurator.app/Contents/MacOS/nwjs
Identifier:      com.nw-builder.betaflight-configurator
Version:         Version 10.3.0 (10.3.0)
Code Type:       X86-64 (Native)
Parent Process:  launchd [187]
Responsible:     nwjs [585]
User ID:         501

Date/Time:       2018-06-24 17:32:44.357 +0200
OS Version:      Mac OS X 10.9.5 (13F1911)
Report Version:  11
Anonymous UUID:  FB7FD1E1-0AC3-A330-8255-6BD27FF51945

Crashed Thread:  0  Dispatch queue: com.apple.main-thread

Exception Type:  EXC_CRASH (SIGABRT)
Exception Codes: 0x0000000000000000, 0x0000000000000000

Application Specific Information:
abort() called

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	0x00007fff93a3d866 __pthread_kill + 10
1   libsystem_pthread.dylib       	0x00007fff9672735c pthread_kill + 92
2   libsystem_c.dylib             	0x00007fff90e35b26 abort + 125
3   com.nw-builder.betaflight-configurator	0x000000010efebd95 main + 325
4   libdyld.dylib                 	0x00007fff99f2b5fd start + 1

Thread 0 crashed with X86 Thread State (64-bit):
  rax: 0x0000000000000000  rbx: 0x00007fff7f2bb310  rcx: 0x00007fff50c14c38  rdx: 0x0000000000000000
  rdi: 0x0000000000000707  rsi: 0x0000000000000006  rbp: 0x00007fff50c14c60  rsp: 0x00007fff50c14c38
   r8: 0x0000000000000040   r9: 0x00007fff50c14b00  r10: 0x0000000008000000  r11: 0x0000000000000206
  r12: 0x00007f872a80b000  r13: 0x00006000000e2800  r14: 0x0000000000000006  r15: 0x0000000000000001
  rip: 0x00007fff93a3d866  rfl: 0x0000000000000206  cr2: 0x00007f872a402000
  
Logical CPU:     0
Error Code:      0x02000148
Trap Number:     133

Binary Images:
       0x10efeb000 -        0x10efebffb +com.nw-builder.betaflight-configurator (Version 10.3.0 - 10.3.0) <855BDBB5-D555-3530-A196-1A58DD4621AD> /Applications/Betaflight Configurator.app/Contents/MacOS/nwjs
    0x7fff6846f000 -     0x7fff684a2887  dyld (239.5) <1D3130FE-FE7E-3C4C-8E74-EB51895B6BA5> /usr/lib/dyld
    0x7fff8d32b000 -     0x7fff8d35afff  com.apple.DebugSymbols (106 - 106) <E1BDED08-523A-36F4-B2DA-9D5C712F0AC7> /System/Library/PrivateFrameworks/DebugSymbols.framework/Versions/A/DebugSymbols
    0x7fff8d43e000 -     0x7fff8d70fffc  com.apple.CoreImage (9.4.1) <7B28C3F3-8EB3-3150-A55D-EBCDBDCA4977> /System/Library/Frameworks/QuartzCore.framework/Versions/A/Frameworks/CoreImage.framework/Versions/A/CoreImage
    0x7fff8d710000 -     0x7fff8daf1ffe  libLAPACK.dylib (1094.5) <7E7A9B8D-1638-3914-BAE0-663B69865986> /System/Library/Frameworks/Accelerate.framework/Versions/A/Frameworks/vecLib.framework/Versions/A/libLAPACK.dylib
    0x7fff8daf2000 -     0x7fff8dd3bfff  com.apple.AddressBook.framework (8.0 - 1371.2) <7DB320AB-4ED4-391E-89CC-125826D94026> /System/Library/Frameworks/AddressBook.framework/Versions/A/AddressBook
    0x7fff8dd3c000 -     0x7fff8dd6cfff  com.apple.IconServices (25 - 25.17) <4751127E-FBD5-3ED5-8510-08D4E4166EFE> /System/Library/PrivateFrameworks/IconServices.framework/Versions/A/IconServices
    0x7fff8dd6d000 -     0x7fff8dd70fff  libCoreVMClient.dylib (58.1) <EBC36C69-C896-3C3D-8589-3E9023E7E56F> /System/Library/Frameworks/OpenGL.framework/Versions/A/Libraries/libCoreVMClient.dylib
    0x7fff8dd71000 -     0x7fff8dd72fff  libunc.dylib (28) <62682455-1862-36FE-8A04-7A6B91256438> /usr/lib/system/libunc.dylib
    0x7fff8dd76000 -     0x7fff8dde3ff1  com.apple.ApplicationServices.ATS (360 - 363.6) <828C2711-4577-3F75-B436-3BDF328DFB11> /System/Library/Frameworks/ApplicationServices.framework/Versions/A/Frameworks/ATS.framework/Versions/A/ATS
    0x7fff8de50000 -     0x7fff8de50fff  com.apple.Accelerate (1.9 - Accelerate 1.9) <509BB27A-AE62-366D-86D8-0B06D217CF56> /System/Library/Frameworks/Accelerate.framework/Versions/A/Accelerate
    0x7fff8de51000 -     0x7fff8de7aff7  libc++abi.dylib (49.1) <21A807D3-6732-3455-B77F-743E9F916DF0> /usr/lib/libc++abi.dylib
    0x7fff8de7b000 -     0x7fff8de93ff7  com.apple.GenerationalStorage (2.0 - 160.3) <64749B08-0212-3AC8-9B49-73D662B09304> /System/Library/PrivateFrameworks/GenerationalStorage.framework/Versions/A/GenerationalStorage
    0x7fff8de94000 -     0x7fff8deb8ff7  libJPEG.dylib (1052) <6836134F-8701-3A6B-82F3-19A0A72CE21C> /System/Library/Frameworks/ImageIO.framework/Versions/A/Resources/libJPEG.dylib
    0x7fff8deb9000 -     0x7fff8def8fff  libGLU.dylib (9.6.5) <7463B411-2DB0-3338-BC8D-403293E2CA34> /System/Library/Frameworks/OpenGL.framework/Versions/A/Libraries/libGLU.dylib
    0x7fff8def9000 -     0x7fff8defafff  liblangid.dylib (117) <9546E641-F730-3AB0-B3CD-E0E2FDD173D9> /usr/lib/liblangid.dylib
    0x7fff8df75000 -     0x7fff8e249fc7  com.apple.vImage (7.0 - 7.0) <D241DBFA-AC49-31E2-893D-EAAC31890C90> /System/Library/Frameworks/Accelerate.framework/Versions/A/Frameworks/vImage.framework/Versions/A/vImage
    0x7fff8e2a8000 -     0x7fff8e3cafff  com.apple.avfoundation (2.0 - 651.12.1) <FDCF1F41-E0BD-3F8C-974A-3F48A0A85B00> /System/Library/Frameworks/AVFoundation.framework/Versions/A/AVFoundation
[... 173 lines cut ...]
__UNICODE                           544K
shared memory                         4K
===========                      =======
TOTAL                             260.5M

Model: Macmini6,2, BootROM MM61.0106.B0A, 4 processors, Intel Core i7, 2.3 GHz, 16 GB, SMC 2.8f1
Graphics: Intel HD Graphics 4000, Intel HD Graphics 4000, Built-In
Memory Module: BANK 0/DIMM0, 8 GB, DDR3, 1600 MHz, 0x0198, 0x393930353432382D3137332E413030472020
Memory Module: BANK 1/DIMM0, 8 GB, DDR3, 1600 MHz, 0x0198, 0x393930353432382D3137332E413030472020
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x10E), Broadcom BCM43xx 1.0 (5.106.98.100.22)
Bluetooth: Version 5.2.7f4 15858, 3 services, 15 devices, 1 incoming serial ports
Network Service: Ethernet, Ethernet, en0
Serial ATA Device: APPLE HDD HTS541010A9E662, 1 TB
USB Device: Hub
USB Device: Hub
USB Device: Keyboard Hub
USB Device: Composite Device
USB Device: Apple Keyboard
USB Device: Hub
USB Device: BRCM20702 Hub
USB Device: Bluetooth USB Host Controller
USB Device: IR Receiver
USB Device: USB Receiver
Thunderbolt Bus: Mac mini, Apple Inc., 23.4
```

furagu, 2018-06-24:
Is there a 10.4.0 build for Mac OS somewhere? I only found Windows and Linux builds on Jenkins.

mikeller, 2018-06-25:
Duplicates #1073.

furagu, 2018-06-28:
Hey @mikeller 
I just tried 10.3.1 on my machine, and it still crashes the same way. I believe it is actually not a duplicate of #1073 . Could be related to Mac OS version (10.9) somehow because it does not crash on another machine running Mac OS 10.11.

mikeller, 2018-06-29:
Looks like this is not a duplicate indeed:

https://www.theinquirer.net/inquirer/news/3034818/chrome-quietly-drops-support-for-mac-os-x-109-mavericks

Since the NW.js toolkit that we are using to build the configurator is based on Chrome, it will only run on platforms supported by Chrome. Looks like support for Mac OS 10.9 got dropped from Chrome, sorry.

furagu, 2018-07-03:
Oh well, finally a REAL reason to update the Mac OS :)

iminchan, 2018-08-08:
For those who can not upgrade beyond 10.9.5 (Mavericks), here is a hack to install High Sierra on older macs. (early 2008~early 2009)

http://dosdude1.com/highsierra/

closed 2018-09-14 by stale[bot]