betaflight/betaflight-configurator #1082: Problem using 10.3 with 3.1.7 and 3.2.3

FIynix, 2018-06-24:
I have an omnibus (F3) with 3.1.7 and an omnibusF4 with 3.2.3.
Using betaflight configurator 10.3 with them caused the quads to become impossible to fly.
   - Quad started to have big vibration / propwash.
   - Sometimes the quad goes full throttle and I need to disarm to stop it.
I found that the Gyro lowpass 1 is turned off and cannot be turned on. (Maybe it's normal ?)
I got the F3 to fly in 3S with very low P and D values, but the feeling was really bad and couldn't get it to fly in 4S (Note : this is a 3in quad).
My 4in quad with omnibusF4 3.2.3 could fly in 4S after lowering P and D, but couldn't get the I term to do it's job correctly and the quad was flying a lot better before 10.3. Note that all I did was open quad in 10.3 to check what's new in 10.3, click save (for no reason, just a reflex) and closed betaflight.

I have now updated my omnibusF4 to 3.4 and will test it tomorrow. If everything goes fine I will do the same with the F3 board and will let you know how it goes.

Maybe there should be a warning not to use 10.3 with version previous to 3.4, or am I the only one with this problem ?

wind0r, 2018-06-24:
Do you have any config dump/diff from befor and after using 10.3? Also could you reproduce the issue with older configurator versions?

Technical there shouldnt be any problem with using a newer configurator. Also I have no idea how that should change anything.

If both dumps are the same it should fly exactly the same.

banditjac, 2018-06-25:
Warning!! I feel your pain, I updated to 3.3, totally blanking the fact the configuator might need to be updated, and I put my quad in the air, flew a stretch about 75 yards, and cooked a brand new Cobra VEK Edition motor. That's a first for me, and I've been building for 5yrs now. So I hang my head in shame lmao. But, FYI or food for thought, don't get too confident. Anywho, just a little share for all. Not complaining, I should know better, and also just want to add a huge thumbs up to the Boris and the betaflight crew, great job guys, love the progress, and appreciate all the hard work! Thank you all!!

seaweeduk, 2018-06-25:
I saw this problem on an older 3.4 build from a few weeks ago. I changed the gyro LPF from my previous value using the latest version of the configurator. The configurator displayed the value, but in the CLI the gyro LPFs were set to 0. 

I upgraded to RC3 and the problem went away. However someone was in slack with the same problem on 3.2.2 so sounds like it's an issue for older builds with the latest configurator. 

I was running on a BFF3 

Here's post from slack

> Ok i took 2 of my quads, they both have 3.3.2 on them,  I hadn't used 10.3.0 before.  I made some changes to Setpoint, saved it.  And now I realize Gryo Lowpass was changed from Normal to Off.  At least thats what it looks like

flint723, 2018-06-25:
@FIynix It isn't good for Gyro Lowpass 1 to be off, for now you'll have to set `gyro_lowpass_hz = 90` (or whatever you had) in CLI.  This seems to allow the value to save.

It's reported that configurator 10.3.0 works fine with 3.4 rc2 and rc3.

flint723, 2018-06-25:
So far reproduced on omnibusf4, 3.3.3

To reproduce:
Flash firmware.  Change no settings.
Enter Pid or Filters tab.  Click save
Click a different tab, then go back to filter tab.  Gyro Lowpass 1 will be disabled.

FIynix, 2018-06-25:
@flint723  I tried to set the gyro in the CLI, but the value wasn't changing in the configurator. So instead of taking the risk of testing it, I updated to 3.4.
Now both of my quad fly perfectly with 3.4 rc3, even better than before the problem.

closed 2018-06-26 by mikeller