betaflight/betaflight-configurator #1086: Files saved from CLI on non-windows contain CRLF (ctrl-M)

jchodakowski, 2018-06-26:
It's the bane of *nix environments - the dreaded ^M - present in all files saved from the CLI button "Save to File".

mikeller, 2018-06-26:
Are there any problems arising from that? It would be easy to fix, but as far as I know nothing is affected by the files being stored in msdos format on all platforms, so nobody has put in the effort to change it.

jchodakowski, 2018-06-28:
Depends on who you ask, I suppose. It's not a breaking bug, but for those of us who live a life of source control and command line tools, the difference in the line termination makes a difference. I happen to keep my flight controller configs in a git repo, so I can track my changes. It's fixable locally with dos2unix, but IMO, if one goes to the trouble of OS support, then at least follow that logic along and use the correct line termination. Not meant to be a snark, just representative of a personal approach. I've forked the project and can track this down if this has (understandably) little priority.

mikeller, 2018-06-28:
@jchodakowski: I still fail to see what actual problems are arising from the configuration being stored in a consistent format across all versions of the configurator. If you want to store your config diffs in a git repo (I do that too by the way), they can be stored / compared just as well in dos format as they can be in unix format.

jchodakowski, 2018-06-29:
The ^M is visible in the editors I use.

AndersHoglund, 2018-06-29:
The BF firmware has no idea what editor or OS a user might use to view files. Using Unix style end-of-line would cause much worse effects and look awful when viewed for example using Windows Notepad. Try it...

mikeller, 2018-06-29:
@jchodakowski:

> The ^M is visible in the editors I use.

This is not a problem per se, just an artefact of the fact that this is a multi-platform project, and files are stored in a portable format.

People familiar with working in multi platform projects are fairly used to seeing these artefacts.

fujin, 2018-06-30:
Regarding 'storing flight controller configs in a git repo', simply set the flight controller configuration repository to convert CRLF's automatically, either via `.gitattributes` on a file-by-file or wildcard basis, or globally for the entire repository. Then when you commit, they'll be rewritten.

https://help.github.com/articles/dealing-with-line-endings/

jchodakowski, 2018-07-03:
In multi-platform projects I've worked, we fix this problem as it applies to each OS -  CRLF may work "enough" but it's not technically correct for anything but Windows.  I'm aware of gitattributes but that's a workaround. 

It's very clear there's zero desire to do anything about this on the code side even tough it's probably not a complicated fix. Nice community.

closed 2018-07-03 by jchodakowski