betaflight/betaflight-configurator #1088: OSD not displaying on screen

Tbeal79, 2018-06-26:
just updated to 10.3 on here, then updated my jbf4 after i did the update to the 3.3.3. set it all up, went out to fly it and i have no osd on the screen at all.
i even come back in to make sure it was all on and everything, all is good as to the setup, but when i plug it in i get nothing on the screen, i have video but it displays nothing at all, i know its there bc i tryed to get into it and you can, but you cant get out of it bc you cant see what you are doing in there, so you have to unplug it to restart it.
not sure what is going on with that

McGiverGim, 2018-06-27:
Start pasting a `diff` please.

Tbeal79, 2018-06-27:
Ok I'll have to get one

mikeller, 2018-06-27:
@Tbeal79: This sounds like a firmware problem, and not a configurator problem. Can you please close this, and open an issue (and supply all of the requested information) on https://github.com/betaflight/betaflight/issues?

Tbeal79, 2018-06-27:
sure thing

closed 2018-06-27 by Tbeal79