betaflight/betaflight-configurator #1094: dterm_notch_cutoff resets to 0 in 10.3.1 when saving pid tab

ghostface, 2018-06-30:
When I save the PID Tab without changing anything it resets my dterm_notch_cutoff to 0 (from the 160).
I am using 10.3.1 and betaflight 3.4 rc3+ the runcam split fix.

I checked by running diff all, going into pid tab, saving, then running diff all again and it suddenly had set dterm_notch_cutoff = 0 there. Tried it a couple of times and was reproducible.

mikeller, 2018-07-01:
I can't reproduce that one. Supplying a `diff` of before and after the reset would be helpful.

mikeller, 2018-07-01:
Looking at your 'before' configuration, it looks like your DTerm notch filter is disabled (`dterm_notch_hz = 0`). This results in both `dterm_notch_hz` and `dterm_notch_cutoff` being set to 0 when saving on the PID tuning tab. What exactly is the problem that this is causing for you?

ghostface, 2018-07-01:
I got no problem, I was just wondering why my diff changed after saving in the PID tab without doing any changes there. 

Thank you for explaining it and looking at the issue so quickly! 
What you said makes perfect sense, I'll close this.

closed 2018-07-01 by ghostface