betaflight/betaflight-configurator #1100: Debian package drops files in the "wrong" location

flexd, 2018-07-02:
Hi there!

I just used the .deb file for the first time, and noticed everything is dumped into /opt/betaflight/betaflight-configurator.  `/usr/bin` or `/usr/local/bin` would be a more suited location for the binary file itself, and `/usr/share` would be an appropriate place for the rest of the files. See https://linux.die.net/man/7/hier 

Since those two folders are already in the PATH of a normal GNU/Linux distro it should just work out of the box. Additionally the deb could drop a .desktop file somewhere perhaps.

Anyway, that's just my opinion as I spent a good few seconds figuring out that the deb dropped the files into /opt :-) You might have a very good reason for using /opt that I'm not aware of. I hope this does not come across as rude, I'm just curious and want to help out.

Thanks for all the good work! Let me know if I can contribute in any way.

McGiverGim, 2018-07-02:
You must have a desktop for created in your menu. It isn't?

flexd, 2018-07-02:
I do not run a normal desktop environment, so there is no normal "start menu" like Gnome/KDE would have. I'm running i3wm, which is a tiled desktop environment. However, I thought I had desktop applications showing up in my launcher menu, perhaps I do not. :-) 

Thanks for the very quick reply! I still think not using `/opt` would be better.

fujin, 2018-07-02:
We abide the Filesystem Hierachy Standard section regarding `/opt/<package>` - http://www.pathname.com/fhs/pub/fhs-2.3.html#OPTADDONAPPLICATIONSOFTWAREPACKAGES

This is a common approach for many "omnibus"/FPM style packaged applications. Debian's packaging guidelines and license compliance requirements would make packaging the Configurator for debian non-free or contrib, etc, time consuming with little to no reward.

In the future if you want to quickly find what files are included in a Debian package, simply use:
`dpkg -L the-package-name`.

closed 2018-07-02 by fujin

flexd, 2018-07-02:
Thanks. I am aware of dpkg -L <package name>; hence it taking just a few seconds to find out. 

Okay then, thanks for getting back to me :-)