betaflight/betaflight-configurator #1107: 3/64 AV engines report malware in 10.3.1 win32 installer

caffeinatedgoat, 2018-07-10:
I realize this is likely due to the way it's packaged but perhaps it could be addressed as it's not a great look.

McGiverGim, 2018-07-11:
Yes, this happens with each new version of the configurator/blackbox that we release. Not too much that we can do to fix it. :-(

mikeller, 2018-07-11:
Yeah, always the same for every release. I suspect the only way we could possibly improve this is if we spent money on a code signing certificate - signed applications seem to be given a lot of negative bias by most virus scanners.
To me, this looks a lot like an elaborate protection money extortion scheme - in particular if applied to an open source project.

frozenskys, 2018-07-11:
Have seen this before with projects I have used NSIS - AV software dosen't seem to like it, and can generate a lot of false positives. The page is a little out of date (2017) but there is a list here : [NSIS False Positives](http://nsis.sourceforge.net/mediawiki/index.php?title=NSIS_False_Positives&oldid=24969)

McGiverGim, 2018-07-11:
I think remember that in one of my tests, the installer generated from windows gave AV warnings, but the same windows installer generated from Linux didn't.

mikeller, 2018-07-12:
@McGiverGim: That's good to know - until now I built the Windows installer under Windows, believing this to be less risky. But if building on linux results in an installer that is less prone to be flagged as a virus, then that seems to be a better option.

@caffeinatedgoat: Can you test this binary to see if it results in less detections: https://drive.google.com/file/d/1DL5MQF6kQFxUeXzboj3dupaFw7Duw8u5/view?usp=sharing

McGiverGim, 2018-07-13:
@mikeller I uploaded your file to virustotal and only 1 engine detected it as a virus, in opposite to the original file, that is detected by 3 engines as a virus.

Here you have the results:
https://www.virustotal.com/#/file/ecd41169ec999ef37b68696265f96f95c7c21efdd98d49103fbf8cba550b0b3f/detection

mikeller, 2018-07-13:
Thanks @McGiverGim. I replaced the released .exe with the one built on linux - if this results in less problems with virus scanners then that's a win.

closed 2018-08-19 by stale[bot]