betaflight/betaflight-configurator #1108: No softserial in GUI

0crap, 2018-07-11:
Using a Matek F405-STD and enable softserial in CLI, it will not show on the Ports page.
By default it shows:
-USB VCP
-UART1
-UART2
-UART3
-UART4
-UART5

McGiverGim, 2018-07-11:
Do you assign it with the `resource` command to some pin? Without this it will not appear.

0crap, 2018-07-11:
No, no "resource" command used.
If I recall correctly it will also not show by simply enable softserial in Configuration tab.

McGiverGim, 2018-07-11:
Usually yes. You must select what pins are you using for the soft-serial feature. If not Betaflight ignores which pins are you using for that. 
Check if that was the problem, and if right, please close this issue ;)

0crap, 2018-07-11:
Thx, I'll do when back at home.

0crap, 2018-07-12:
Works exactly as you said.
My misunderstanding.
Thx!

closed 2018-07-12 by 0crap