betaflight/betaflight-configurator #1111: Copy rateprofile not working correctly

atshort, 2018-07-14:
In configurator 10.3.1 on a Mac, on the PID tuning page, if I use the "copy rateprofile values" button while in profile 2 and I say  "Copy values from current rateprofile to" rateprofile 3, what happens is rateprofile 1 is copied to rateprofile 3.

mikeller, 2018-07-14:
@atshort: At least in your statement you are mixing 'profile' and 'rateprofile'. These are different and independent things. Could it be that you are mixing this up?

atshort, 2018-07-14:
Sorry, I meant rateprofile 2. All of the above took place in profile 1. I just corrected it.

mikeller, 2018-07-14:
@basdelfos: Do you have time to look into this?

richard-scott, 2018-07-16:
In my experience with this you need to click "save" straight after copying a profile. You need to click save before looking at the target profile, otherwise the board settings are loaded for that target profile from the FC and it "looks" like the function didn't work... Maybe the copy feature could be setup to save automatically?

basdelfos, 2018-07-16:
I will have to look into this issue in the coming days.
From what I remember, I think that Richard is right. The function copies the rate-profile struct from one profile to the other profile. Hitting save will store the profile into the EEPROM. But I will have to dig into this to confirm this is the case.

atshort, 2018-07-16:
I am aware of the need to click save. Here are the steps to reproduce what I believe is the problem:

- wipe and flash flight controller
- in rateprofile 1, change something (I changed roll srate to 0.71), and click save
- switch to rateprofile 2, and click save
- click "Copy rateprofile values", and select from current rateprofile to rateprofile 3, then click save

I believe this should copy rateprofile 2 to rateprofile 3, but what I see is it copied rateprofile 1 to rateprofile 3.

smoriarty21, 2018-07-16:
@basdelfos @mikeller If you guys are busy and want me to take a crack at this one let me know.  I'd be happy to jump in and try to figure it out

mikeller, 2018-07-17:
@smoriarty21: Help is always appreciated.

@basdelfos: My suspicion is that this might be caused by interference with a rate profile switch set up on an adjustment range. If this is the case, and there is a valid RX signal, then the switch position is authoritative for the active rate profile on the flight controller, and the rate profile cannot be changed programatically by the configurator.

atshort, 2018-07-17:
Although I first noticed the problem in a working quad, the steps I outlined above to reproduce the problem were carried out on a bare flight controller (Bardwell F4), not connected to anything except USB. There's no RX in the picture, and as I mentioned I wiped the board and did a fresh install before testing so there are no adjustment ranges configured. I literally just performed the three steps outlined above on a freshly wiped/installed board.

basdelfos, 2018-07-28:
Sorry, I didn't found any time to take a look at this. All busy working and preparing for vacation. The next few weeks I can not fix this. I someone else sees time to fix this, please be my guest.

smoriarty21, 2018-07-29:
@mikeller Sounds good, Ill jump on this one after rc smoothing

smoriarty21, 2018-08-21:
@mikeller @etracer65 @McGiverGim 

I fixed the issue of the wrong rate profile being copied. While testing I noticed that profile 6 would not allow anything to be copied to it, dug through the code and put together a fix that worked on my LUXF4OSD.  This is my first time messing with the main betaflight repo so please excuse any mistakes :)

https://github.com/betaflight/betaflight/pull/6612
https://github.com/betaflight/betaflight-configurator/pull/1176

tavdog, 2018-09-11:
I'm seeing this bug on my Matek F405 , BF 3.5.1, mac configurator 10.4.0. When trying to copy rate profile #1 to rate profile #2, nothing happens. I've tried a save before going to rate profile #2 but the default rates persist.  This board is un upgrade from 3.3.3 to 3.5.1 with settings moved over via diff.

McGiverGim, 2019-04-03:
It seems fixed by #1176 and betaflight/betaflight#6612. I close this.

closed 2019-04-03 by McGiverGim