betaflight/betaflight-configurator #1117: What is the APIs to test the motors when mocking pitch/roll/yaw+throttle

weixingsun, 2018-07-17:
Dear Experts,

I am building my own copter, but I cannot find any document about testing the motors without a remote control, please let me know where is the codes to control the Throttle + Pitch/Roll/Yaw
Or give me a hint to develop it , thanks

mikeller, 2018-07-17:
Open configurator, click 'Connect', click 'Motors' tab, click switch next to 'I understand the risks', slide 'Master' slider up, watch motors turn.

weixingsun, 2018-07-17:
@mikeller thanks for your reply, I know this page, and am testing motors as you mentioned, 
it looks good, but I am finding the "combined" movements like yaw/pitch/roll, just like what we do from a remote control.

weixingsun, 2018-07-17:
Still wondering if there is a way to mocking the remote control signals

McGiverGim, 2018-07-18:
In the receiver tab, there is an MSP virtual command, I don't know if it will work... you can try it but with caution.

weixingsun, 2018-07-18:
@McGiverGim Thanks for your reply, I didn't find the MSP virtual command feature in the receiver tab,
there is a page shows the R/P/Y/T and AUX bars but without input fields, please help

McGiverGim, 2018-07-18:
I don't remember if they move with keys or with the mouse. I think remember they are keys (shortcuts). But you must first agree to the message that appears.

weixingsun, 2018-07-18:
[image]
This is my screenshot, I tried key AERT on my keyboard, no luck. this should be protocol of different RC not the keyboard.

McGiverGim, 2018-07-18:
You must activate the MSP RX Receiver in the configuration tab. Then you will see a new button in the bottom right of this screen. Or something similar.

weixingsun, 2018-07-18:
@McGiverGim  Yes, I followed your step and open the control stick page, I can control the AUX signals now. 
But the Roll/Pitch/Yaw/Throttle are still unable to operate on my Macbook, I will try windows version later. 
Thanks a lot

weixingsun, 2018-07-20:
I can use the motor tab to control the speed, but I cannot change my motors from the control stick page under receiver tab, the only effect is the preview animation in the bottom of the page. 
Do I need to enable any configuration?

McGiverGim, 2018-07-20:
I'm not too sure... maybe in the new versions you can't mount the FC while connected to the configurator. Try to go to the motors tab and activate the arming and then try again.

mikeller, 2018-07-21:
@weixingsun: You need to enable motor control in the Motors tab in order for arming to be possible.

weixingsun, 2018-07-21:
OK, but for whatever reason, my motors only run under motors tab, no luck under receiver tab. 
My FC: Omnibus F4v3
Betaflight-Configurator: 10.3.1 mac
anything I can do ?

mikeller, 2018-07-21:
Looks like your RX isn't working properly. `MSP` will go away when you enable the motors in the motor tab, and `CLI` is shown because you are in CLI.

weixingsun, 2018-07-21:
@mikeller I upgraded FC to the latest version 3.4.0, then I see the value changed to "RXLOSS CLI"
without MSP, but I still cannot give it a run

mikeller, 2018-07-21:
@weixingsun: Again, your RX is not working. Set up a working RX and it will work.

McGiverGim, 2018-07-23:
If I'm not wrong, you must open the MSP RX input panel, that is a standalone window, and then you can change to the motors tab, activate it, and use it.