betaflight/betaflight-configurator #1129: KAKUTE F4v2 no baro - Configurator 10.3.1 BF 3.4.0 release

RotorVulura, 2018-07-23:
Hi there !
I can see the baro in the sensors tab working OK, but it does not turn on when a TX switch is assigned to (in "modes" tab I can see the switch works properly, but the baro doesn't change its state to yellow).
In BF 3.5 with Configurator 10.4.0 there is not "baro" feature in the "modes" tab. Totally disappeard. Again, I can see the baro working properly in the "sensors" tab.
Thx in advance :)

McGiverGim, 2018-07-23:
The baro mode was removed in BF 3.4. In the next configurator it will be removed too.

RotorVulura, 2018-07-23:
D’oh ! May I ask why and where is the documentation that talking about it and other “features”. Just to know ... I don’t like to open new “issue” that you call “feature”. It’s a waste of time. Thx for your answer.

McGiverGim, 2018-07-23:
All the navigation modes were removed because they simply not work (baro and gps modes). A new gps rescue mode was added that works.
Maybe in the release notes of the version appears but I'm not sure.

RotorVulura, 2018-07-23:
Ok, here it is: https://github.com/betaflight/betaflight/pull/5764
So, I need a gps module now ... I was happy in the past triggering baro and, why not, angle mode and just bring  back the quad in a confort zone if the vtx signal get lost. That’s it ... I’m going to close this issue and remove the baro from the FC. It will be a bit lighter :-)

closed 2018-07-23 by RotorVulura